OpenDroneMap/ODM #666: Could not find a package configuration file provided by "VTK"

amiteshsah, 2017-09-04:
I tried to install Opendronemap in my virtual machine (16.04)  by cloning the repository from github and followed the instructions carefully as described in the README. This is the error i am getting 

By not providing "FindVTK.cmake" in CMAKE_MODULE_PATH this project has
  asked CMake to find a package configuration file provided by "VTK", but
  CMake did not find one.

  Could not find a package configuration file provided by "VTK" with any of
  the following names:

    VTKConfig.cmake
    vtk-config.cmake

  Add the installation prefix of "VTK" to CMAKE_PREFIX_PATH or set "VTK_DIR"
  to a directory containing one of the above files.  If "VTK" provides a
  separate development package or SDK, be sure it has been installed.

This is another error at the end of the installation
Linking CXX executable ../../bin/odm_orthophoto
[100%] Built target odm_orthophoto
make[2]: *** [modules/odm_meshing/CMakeFiles/odm_meshing.dir/src/main.cpp.o] Error 1
make[2]: *** Waiting for unfinished jobs....
make[2]: *** [modules/odm_meshing/CMakeFiles/odm_meshing.dir/src/OdmMeshing.cpp.o] Error 1
make[1]: *** [modules/odm_meshing/CMakeFiles/odm_meshing.dir/all] Error 2
make[1]: *** Waiting for unfinished jobs....
Linking CXX executable ../../bin/odm_texturing
[100%] Built target odm_texturing
Linking CXX executable ../../bin/odm_georef
[100%] Built target odm_georef
Linking CXX executable ../../bin/odm_25dmeshing
[100%] Built target odm_25dmeshing
make: *** [all] Error 2
Configuration Finished

I am attaching the log file so that you can have better insight  into the cause of the problem
[odm_install.txt](https://github.com/OpenDroneMap/OpenDroneMap/files/1273551/odm_install.txt)

dakotabenjamin, 2017-09-05:
The error in the log is here: 
```
../lib/libpdal_base.so.5.1.0: undefined reference to `Json::Value::operator=(Json::Value)'
collect2: error: ld returned 1 exit status
make[5]: *** [bin/pdal] Error 1
make[4]: *** [apps/CMakeFiles/pdal.dir/all] Error 2
make[4]: *** Waiting for unfinished jobs....
[ 98%] Built target pdal_plugin_writer_pgpointcloud
[ 98%] Built target pdal_plugin_filter_hexbin
make[3]: *** [all] Error 2
make[2]: *** [pdal/stamp/pdal-build] Error 2
make[1]: *** [CMakeFiles/pdal.dir/all] Error 2
make[1]: *** Waiting for unfinished jobs....
```

amiteshsah, 2017-09-06:
@dakotabenjamin  Do you have any suggestion how to fix that? I tried multiple times both locally and in server, it is giving me the same error.

dakotabenjamin, 2017-09-14:
Have you tried using docker?

amiteshsah, 2017-09-18:
I was doing without docker .I will try to do with the docker and update you.

closed 2018-03-28 by dakotabenjamin