OpenDroneMap/ODM #670: Missing opensfm on the current master branch

hy9be, 2017-09-12:
I clone the current repo and try to build the master branch via the approach mentioned [here](https://github.com/OpenDroneMap/OpenDroneMap/wiki/Installation). But the SuperBuild/src/opensfm folder is empty after the installation, thus ODM cannot run.

Anyone see the same issue? Thanks!

dakotabenjamin, 2017-09-13:
Can you run `git pull` and then `configure.sh reinstall`?

hy9be, 2017-09-14:
Yes, I tried that but still see the empty folder. So it is okay for you guys right?

dakotabenjamin, 2017-09-14:
I sometimes run into this. How big is your processor?

hy9be, 2017-09-15:
I am using a 8 cpu machine. It is the aws large CPU optimized instance.

rionlerm, 2017-09-15:
@dakotabenjamin same issue with me. Could it be related to this:

hy9be, 2017-09-18:
@rionlerm I am not sure if ours are the same issue. Does your configure.sh start with ``#!/bin/bash``? Or maybe you can try ``bash configure.sh reinstall``.

rionlerm, 2017-09-18:
Hi @hy9be, my configure.sh is run via the latter code you mentioned. Your install approach I used in the past to install ODM a while back. Since then follow new approach by just reinstalling.

closed 2018-01-02 by hy9be