OpenDroneMap/ODM #673: run ODM with Docker in Windows

danng007, 2017-09-21:
Hi,

I'm a new user for this. After all install, I got an error: Not enough supported images in /code/images. Anyone can help please?

I created a folder called 'images' in current working directory, and downloaded some images from https://github.com/OpenDroneMap/odm_data. 

my docker running command is : docker run -it --rm -v $(pwd)/images:/code/images -v $(pwd)/odm_orthophoto:/code/odm_orthophoto -v $(pwd)/odm_georeferencing:/code/odm_georeferencing opendronemap/opendronemap. Then the error will appear.

Thanks for anyone's help.
Best,
Wantao

danng007, 2017-09-21:
The full print after enter the command is:
[INFO]    Initializing OpenDroneMap app - Thu Sep 21 16:05:50  2017
[INFO]    Running ODM Load Dataset Cell
[DEBUG]   Loading dataset from: /code/images
[DEBUG]   /code/images
[ERROR]   Not enough supported images in /code/images
[INFO]    OpenDroneMap app finished - Thu Sep 21 16:05:50  2017

PeterSprague, 2017-09-24:
I don't work in Windows, but first thing I would check is your permissions.

I noticed you had asked for help on gitter.  My take on this issue is that it is not a ODM code issue, but specific to the user environment.  That is what the Gitter help page is for.

closed 2017-09-25 by dakotabenjamin