OpenDroneMap/ODM #674: odm_extract_utm does not parse altitude correctly

dakotabenjamin, 2017-09-21:
Thanks @fredlllll for helping diagnose this bug

In https://github.com/OpenDroneMap/OpenDroneMap/blob/master/modules/odm_extract_utm/src/UtmExtractor.cpp#L310 odm_utm_extract is looking for the jhead output with "GPS Altitude :" but this is not always how image exif is saved. See: 
[image]
vs:
[image]

This is both an issue with odm_utm_extract AND jhead. We should find a new library for reading image exif into C++ and update the module. Options include ImageMagick's identify script, exiftool (which is slow and depends on perl), libexif, or [Exiv2](http://www.exiv2.org).

dakotabenjamin, 2017-09-21:
exiv2 dependencies:
```
Feature                     Package   Configure options
--------------------------  --------  ----------------------------
PNG image support           zlib      --without-zlib
                                      --with-zlib=DIR
Native language support     gettext   --disable-nls
Characterset conversions    libiconv  --without-libiconv-prefix
                                      --with-libiconv-prefix[=DIR]
XMP support                 expat     --disable-xmp
                                      --with-expat=DIR

zlib         http://zlib.net/
gettext  *)  http://www.gnu.org/software/gettext/
libiconv *)  http://www.gnu.org/software/libiconv/
expat        http://expat.sourceforge.net/
```

fredlllll, 2017-09-21:
dependencies for imagemagick here https://github.com/ImageMagick/ImageMagick/blob/master/Install-unix.txt under the "Delegate Configuration"

dakotabenjamin, 2017-09-21:
a native c++ library is probably more ideal anyways. see: http://exiv2.org/examples.html#example1

fredlllll, 2017-09-21:
looks much cleaner than the current parsing using a temporary file and the sketchy string processing

KommandorKeen, 2017-09-25:
Just pulled your fix and ran it. 
In the opensfm directory the data in the individual exif data files matches the RTK data I inserted into the header using EXIFtool :-).  Just processing the images now, but this is a great step forward as the aircraft is a Phantom4 with an Emlid Reach kinematic GPS post-processed.

dakotabenjamin, 2017-09-25:
Merged in master

closed 2017-09-25 by dakotabenjamin