OpenDroneMap/ODM #676: Canon G10

yjmenezes, 2017-09-26:
I wanna test a Canon G10 with OSM.
the firmware is 1.0.3.0
has someone used this camera and can give me some directions about shot control thru  Pixhawk or APM ?  ( cable, fw update... )

thx

LucasMM86, 2017-09-26:
Hello, i use canon s100 with pixhawk.
Therfore, CHDK is installed on the camera (http://chdk.wikia.com/wiki/CHDK). Seems that the G10 is supported (http://chdk.wikia.com/wiki/G10), take care about the firmware version. A friend of mine did the installation so i can not give you more tips about this.
If you use pixhawk you will need some electronics on the cable in order to convert 3V from the controller to 5V. Here is how to do that: http://ardupilot.org/plane/docs/common-apm-to-chdk-camera-link-tutorial.html
Then you need to configure the controller to trigger the camera. If you use Mission Planner: http://ardupilot.org/plane/docs/common-camera-shutter-with-servo.html#common-camera-shutter-with-servo
Another option is to avoid the cable and let the camera to trigger automatically each x seconds, lets say 5 seconds. This is suported by CHDK.
regards

dakotabenjamin, 2017-09-26:
The advantage to using the trigger cable is the ability to use the drone telemetry to geotag the photos. It is easily done inside Mission Planner

LucasMM86, 2017-09-26:
It is also possible to geotagg using time triggering in Mission Planner!
In my experience, the main advantage of using the trigger cable is that the lens can be retracted inmediatly at the end of the mission, so you can land with the lens retracted and you dont scratch the lens, specially using fixed wing drones.

yjmenezes, 2017-09-26:
thx @LucasMM86 and @dakotabenjamin 

my doubt is if G10 fw 1.0.3.0 is CHDK compatible.
the chdk wiki shows G10 fw  1.0.3b  and mine is 1.0.3.0

http://chdk.wikia.com/wiki/G10

thx

closed 2018-03-28 by dakotabenjamin