OpenDroneMap/ODM #678: Missing Checkerboard link

Ntweat, 2017-09-28:
Hi 
I built opendrone with ORB SLAM, but the checkerboad link is not working.
Where else can I find it?

pierotofy, 2017-09-28:
Does this work? http://docs.opencv.org/2.4/_downloads/pattern.png

Could you also point out where the broken link is exactly?

Ntweat, 2017-09-28:
@pierotofy  Thanks 

Links are broken on https://github.com/OpenDroneMap/OpenDroneMap/wiki/Reconstruction-from-Video 

All the dropbox links are broken

pierotofy, 2017-09-28:
Mm, @smathermather  are you dropbox user `2801164`? 😄

smathermather, 2017-09-29:
Noap. But I bet @paulinus is.

Ntweat, 2017-10-02:
@pierotofy @smathermather  can you give me the config.yaml and the castle video files? 
I tried running opendrone sfm on my video it is not able to track the camera or features.

dakotabenjamin, 2018-03-28:
@Ntweat sorry for the late reply- I've fixed some of the links but I don't have access to the video files. I hope that helps a little.

dakotabenjamin, 2018-03-28:
This issue was moved to http://community.opendronemap.org/t/missing-checkerboard-link/380

closed 2018-03-28 by dakotabenjamin