OpenDroneMap/ODM #682: Generating texture patches Killed

YingyingXie16, 2017-10-06:
I'm running opendronemap through docker on windows10.

[image]

I tried a few times, but always got empty folders of odm_orthophoto and odm_georeferencing, and a folder of odm_texturing with three files: odm_textured_model_data_costs, odm_textured_model.conf, and odm_textured_model_labeling.vec.

The processing showed:
[image]

I'm not sure what's wrong. Please help. Thanks.

dakotabenjamin, 2017-10-06:
Can you share the odm_meshing/odm_mesh.ply?

YingyingXie16, 2017-10-06:
Yeah, but could you help me find the odm_meshing/odm_mesh.ply? I'm not sure where to find it...thanks.

bingzhang, 2017-10-11:
Hello,

I have the same issue on running on the sample dataset ``Download a sample dataset from here (about 550MB)''

I use docker ce17 on mac.

`docker run -it --rm -v /localpath/odm_data_aukerman-master/images/:/code/images opendronemap/opendronemap`

```
	0000	Increase of energy - stopping optimization
	Took: 54.74s
Generating texture patches:
	Running... Killed
Traceback (most recent call last):
  File "/code/run.py", line 47, in <module>
    plasm.execute(niter=1)
  File "/code/scripts/mvstex.py", line 124, in process
    '{keepUnseenFaces}'.format(**kwargs))
  File "/code/opendm/system.py", line 34, in run
    raise Exception("Child returned {}".format(retcode))
Exception: Child returned 137
```

YingyingXie16, 2017-10-11:
@dakotabenjamin I rerun the process and output the odm-mesh.ply file today. I also opened this file in MeshLab. It seems that it's ok as it does show the right landscape captured in my test photos, just some trees around a parking lot with cars and a few buildings. I still do not know what's wrong with the killed texturing process. 

Please check the odm_meshing/odm_mesh.ply in the attached zip file. Thanks.

[odm_meshing.zip](https://github.com/OpenDroneMap/OpenDroneMap/files/1377154/odm_meshing.zip)

dakotabenjamin, 2017-10-13:
Can you try running without the --mesh-size option? It may be decimating too much.

bingzhang, 2017-10-14:
@dakotabenjamin , please see my post, I did not use --mesh-size option, but still get the same error.

YingyingXie16, 2017-10-17:
@dakotabenjamin OK, I ran it without the --mesh-size option, and it proceeded a bit but still got Blending texture patches killed...Please help.
`Generating texture patches:
        Running... done. (Took 33.747s)
        4521 texture patches.
Running global seam leveling:
        Create matrices for optimization...  done.
        Lhs dimensionality: 105378 x 105378
        Calculating adjustments:
                Color channel 0: CG took 112 iterations. Residual is 9.78849e-05
                Color channel 2: CG took 113 iterations. Residual is 9.8304e-05
                Color channel 1: CG took 114 iterations. Residual is 9.91845e-05
                Took 0.801 seconds
        Adjusting texture patches 100%... done. (Took 100.08s)
Running local seam leveling:
        Blending texture patches 21%...Killed
Traceback (most recent call last):
  File "/code/run.py", line 47, in <module>
    plasm.execute(niter=1)
  File "/code/scripts/mvstex.py", line 124, in process
    '{keepUnseenFaces}'.format(**kwargs))
  File "/code/opendm/system.py", line 34, in run
    raise Exception("Child returned {}".format(retcode))
Exception: Child returned 137`

dakotabenjamin, 2017-10-17:
OK both of you please tell me how much memory is allocated to docker, and then increase if it's only like 4GB

bingzhang, 2017-10-17:
Thanks, @dakotabenjamin! 
By increasing the memory, it works perfectly. Maybe a further question, the memory will be related to the size of datasets? What is your suggestion of the minimum memory.

dakotabenjamin, 2017-10-18:
Yeah more images => more memory overhead. I suggest 8 for bare minimum

closed 2018-03-28 by dakotabenjamin