OpenDroneMap/ODM #683: How to do only orthophotos in ODM?

Danicar7, 2017-10-10:
I want to do an orthotography of a terrain to get its NDVI, but ODM generates the textures and map of points, I would like to know if there is any way to generate only the orthophoto, this to optimize the analysis and not take so long to generate the orthophoto.
I would also like to know if there is an equal tool in ODM that can compare 2 orthophotos and generate a single one to get the NDVI.

smathermather, 2017-10-11:
Hi @Danicar7 -- the short answer is no. The process is a full photogrammetric toolchain, which requires the generation of intermediate 3D portions in order to correct for all factors. In the FOSS arena, I know of no software that can do what you propose, which would require feature matching to be combined with thin plate spline corrections between images.

I am very interested in this question as well, and long-term hope to build a tool that can do this, but for now, this is definitely not a function of ODM.

fredlllll, 2017-10-25:
the orthophoto is generated from the georeferenced mesh. thats why it has to be generated beforehand.

PeterSprague, 2017-10-26:
"I would also like to know if there is an equal tool in ODM that can compare 2 orthophotos and generate a single one to get the NDVI."

Not in ODM, but if orthophotos align to the resolution you need, use QGIS raster calculator.  Can also align your orthos to known georeference points.