OpenDroneMap/ODM #684: ODM Docker - File "/code/opendm/system.py", line 53 - [Errno 13] Permission denied: '/code/benchmark.txt'

timisonyourside, 2017-10-16:
Am running the latest version of Docker OpenDroneMap and am getting the following error (I think right after OpenSFM starts) with this command - PYTHONPATH=/code/SuperBuild/install/lib/python2.7/dist-packages /code/SuperBuild/src/opensfm/bin/export_bundler /code/opensfm: 

  File "/code/run.py", line 47, in <module>
    plasm.execute(niter=1)
  File "/code/scripts/opensfm.py", line 162, in process
    system.benchmark(start_time, tree.benchmarking, 'OpenSfM')
  File "/code/opendm/system.py", line 53, in benchmark
    with open(benchmarking_file, 'a') as b:
IOError: [Errno 13] Permission denied: '/code/benchmark.txt'

Here is the command my script built to run ODM (I was having container permission problems, so I pre-created all these mapped volumes in the local file system ahead of running the command): docker run -u1000:1000 -v/etc/group:/etc/group:ro -v/etc/passwd:/etc/passwd:ro -v/home/user1/dev/sitemon/temp/a642f9060c37034f871ad5411ae495f2/original:/code/original -v/home/user1/dev/sitemon/temp/a642f9060c37034f871ad5411ae495f2/images:/code/images -v/home/user1/dev/sitemon/temp/a642f9060c37034f871ad5411ae495f2/opensfm:/code/opensfm -v/home/user1/dev/sitemon/temp/a642f9060c37034f871ad5411ae495f2/pmvs:/code/pmvs -v/home/user1/dev/sitemon/temp/a642f9060c37034f871ad5411ae495f2/odm_orthophoto:/code/odm_orthophoto -v/home/user1/dev/sitemon/temp/a642f9060c37034f871ad5411ae495f2/odm_georeferencing:/code/odm_georeferencing -v/home/user1/dev/sitemon/temp/a642f9060c37034f871ad5411ae495f2/odm_texturing:/code/odm_texturing -v/home/user1/dev/sitemon/temp/a642f9060c37034f871ad5411ae495f2/images_resize:/code/images_resize -v/home/user1/dev/sitemon/temp/a642f9060c37034f871ad5411ae495f2/odm_meshing:/code/odm_meshing -v/home/user1/dev/sitemon/temp/a642f9060c37034f871ad5411ae495f2/odm_mesh:/code/odm_mesh --rm -t opendronemap/opendronemap --force-ccd 17.3 --opensfm-processes 2 --resize-to 3200 --min-num-features 10000 --pmvs-min-images 5 --mesh-size 850000 --mesh-octree-depth 10 --time --end-with odm_georeferencing

I didn't see any current issues with benchmark.txt in my search of OpenDroneMap issues. I'm not sure how to interpret "Permission denied: '/code/benchmark.txt'" when /code/ seems to be owned by the Docker container itself. Even if I had to map /code/ to my local file system, what would I map it to? I am running on Ubuntu 14.

Any ideas?  Thanks!

dakotabenjamin, 2017-10-16:
It's a volume mounting problem with docker. You can solve it by adding
```
-v /home/user1/dev/sitemon/temp/a642f9060c37034f871ad5411ae495f2/benchmark.txt:/code/benchmark.txt 
```

timisonyourside, 2017-10-16:
Thanks for looking that this, Dakota. I'm assuming that OpenSFM was expecting to read content from the benchmark.txt file in the container file system at /core, so I don't expect that manually creating an empty file locally and mapping it will help if that is the case. Should I try it anyway or wait to see? Thanks.

dakotabenjamin, 2017-10-16:
OK so I looked and I didn't accidentally delete the code that creates the benchmark file (it's in scripts/odm_app.py)

OpenSfM is trying to open the file to append it's benchmark data to it. I was thinking if you were having permissions issues on the other directories, then you could pre-create the benchmark.txt file and it would work as well.

timisonyourside, 2017-10-16:
Ok, I'll create an empty benchmark.txt file, map it to the container path and give that a try. 

It does seem that this file should be in /code/opensfm instead of in /code, since you'd expect opensfm-specific files would be grouped under /code/opensfm.

dakotabenjamin, 2017-10-16:
It belongs in the base dir because it records the timings of each step, not just opensfm

timisonyourside, 2017-10-16:
Ah, didn't know it was used more broadly.

Oh, so I could maybe try to more generally map the project directory directly to /code, ala:  -v /home/user1/dev/sitemon/temp/a642f9060c37034f871ad5411ae495f2:/code for any other file accesses at the /code level?

timisonyourside, 2017-10-17:
Hi, Dakota, unfortunately, still the same error after I pre-allocated the benchmark.txt and mapped it to /core/benchmark.txt. When I touch a file, it's access mode is created as 755. Thought that might be an issue if the mode wasn't 777, so I made sure all pre-allocations had a mode of 777. No luck.

Here is the error from the 777 run: File "/code/run.py", line 47, in <module>
    plasm.execute(niter=1)
  File "/code/scripts/opensfm.py", line 162, in process
    system.benchmark(start_time, tree.benchmarking, 'OpenSfM')
  File "/code/opendm/system.py", line 53, in benchmark
    with open(benchmarking_file, 'a') as b:
IOError: [Errno 13] Permission denied: '/code/benchmark.txt'

Here is my docker command: docker run -u1000:1000 -v/etc/group:/etc/group:ro -v/etc/passwd:/etc/passwd:ro -v/home/user1/dev/sitemon/temp/a642f9060c37034f871ad5411ae495f2/original:/code/original -v/home/user1/dev/sitemon/temp/a642f9060c37034f871ad5411ae495f2/images:/code/images -v/home/user1/dev/sitemon/temp/a642f9060c37034f871ad5411ae495f2/opensfm:/code/opensfm -v/home/user1/dev/sitemon/temp/a642f9060c37034f871ad5411ae495f2/pmvs:/code/pmvs -v/home/user1/dev/sitemon/temp/a642f9060c37034f871ad5411ae495f2/odm_orthophoto:/code/odm_orthophoto -v/home/user1/dev/sitemon/temp/a642f9060c37034f871ad5411ae495f2/odm_georeferencing:/code/odm_georeferencing -v/home/user1/dev/sitemon/temp/a642f9060c37034f871ad5411ae495f2/odm_texturing:/code/odm_texturing -v/home/user1/dev/sitemon/temp/a642f9060c37034f871ad5411ae495f2/images_resize:/code/images_resize -v/home/user1/dev/sitemon/temp/a642f9060c37034f871ad5411ae495f2/odm_meshing:/code/odm_meshing -v/home/user1/dev/sitemon/temp/a642f9060c37034f871ad5411ae495f2/odm_mesh:/code/odm_mesh -v/home/user1/dev/sitemon/temp/a642f9060c37034f871ad5411ae495f2/benchmark.txt:/codebenchmark.txt --rm -t opendronemap/opendronemap --force-ccd 17.3 --opensfm-processes 2 --resize-to 3200 --min-num-features 10000 --pmvs-min-images 5 --mesh-size 850000 --mesh-octree-depth 10 --time --end-with odm_georeferencing

Any ideas?

I ran the docker commend using the uid/gid of my user1 user, but it didn't seem to make a difference because files were still getting created by root - that's why I was pre-allocating items.

It seems as though things would work much better if the container was ran with the same uid that runs the command. Is there some other way to affect the uid that executes the docker command? Maybe create the image/container using the non-root id that runs the command?

dakotabenjamin, 2017-10-17:
Have you added your user to the "docker" group?

timisonyourside, 2017-10-17:
Retried the command with the missing slash: docker run -u1000:1000 -v/etc/group:/etc/group:ro -v/etc/passwd:/etc/passwd:ro -v/home/user1/dev/sitemon/temp/a642f9060c37034f871ad5411ae495f2/original:/code/original -v/home/user1/dev/sitemon/temp/a642f9060c37034f871ad5411ae495f2/images:/code/images -v/home/user1/dev/sitemon/temp/a642f9060c37034f871ad5411ae495f2/opensfm:/code/opensfm -v/home/user1/dev/sitemon/temp/a642f9060c37034f871ad5411ae495f2/pmvs:/code/pmvs -v/home/user1/dev/sitemon/temp/a642f9060c37034f871ad5411ae495f2/odm_orthophoto:/code/odm_orthophoto -v/home/user1/dev/sitemon/temp/a642f9060c37034f871ad5411ae495f2/odm_georeferencing:/code/odm_georeferencing -v/home/user1/dev/sitemon/temp/a642f9060c37034f871ad5411ae495f2/odm_texturing:/code/odm_texturing -v/home/user1/dev/sitemon/temp/a642f9060c37034f871ad5411ae495f2/images_resize:/code/images_resize -v/home/user1/dev/sitemon/temp/a642f9060c37034f871ad5411ae495f2/odm_meshing:/code/odm_meshing -v/home/user1/dev/sitemon/temp/a642f9060c37034f871ad5411ae495f2/odm_mesh:/code/odm_mesh -v/home/user1/dev/sitemon/temp/a642f9060c37034f871ad5411ae495f2/benchmark.txt:/code/benchmark.txt --rm -t opendronemap/opendronemap --force-ccd 17.3 --opensfm-processes 2 --resize-to 3200 --min-num-features 10000 --pmvs-min-images 5 --mesh-size 850000 --mesh-octree-depth 10 --time --end-with odm_georeferencing

Got a slightly different path but the same error: File "/code/run.py", line 40, in <module>
    app = ODMApp(args=args)
  File "/code/scripts/odm_app.py", line 26, in __init__
    ecto.BlackBox.__init__(self, *args, **kwargs)
  File "/code/SuperBuild/install/lib/python2.7/dist-packages/ecto/blackbox.py", line 258, in __init__
    self.__configure(**kwargs)
  File "/code/SuperBuild/install/lib/python2.7/dist-packages/ecto/blackbox.py", line 516, in __configure
    self.configure(self.__params, self.__inputs, self.__outputs)
  File "/code/scripts/odm_app.py", line 95, in configure
    os.remove(tree.benchmarking)

Yes, user1 is in the docker group: 
uid=1000(user1) gid=1000(user1) groups=1000(user1),999(docker)

timisonyourside, 2017-10-18:
Any other ideas why I'm seeing this permissions error, Dakota?

This site may be a solution to run the container using the same UID/GID as the user executing the docker command, but only in development and the dockerfile would need to be updated before the build - is this a possible approach? https://github.com/boxboat/fixuid

PeterSprague, 2017-10-18:
What is your host OS?

This is all I use for a docker cmd: docker run -it -v $(pwd)/images/images_jpg:/code/images:Z -v $(pwd)/images_resize:/code/images_resize:Z -v $(pwd)/opensfm:/code/opensfm:Z -v $(pwd)/odm_meshinging:/code/odm_meshing:Z -v $(pwd)/odm_texturing:/code/odm_texturing:Z -v $(pwd)/odm_georeferencing:/code/odm_georeferencing:Z -v $(pwd)/odm_orthophoto:/code/odm_orthophoto:Z -v $(pwd)/settings.yaml:/code/settings.yaml:Z --rm gei/opendronemap:july2017  --force-focal 3.97 --force-ccd 4.63104

timisonyourside, 2017-10-18:
Ubuntu 14.04

I'm not familiar with the 'Z' suffix on the volume mapping command.

Is this a docker command used directly for Docker ODM or for WebODM?

PeterSprague, 2017-10-18:
Z parameter is for Fedora when using Docker volumes.  I'm only using ODM from the cli w/ Fedora 26 host, Docker container built with slightly edited ODM dockerfile

timisonyourside, 2017-10-18:
Thanks, Peter, and you are not seeing the /code/benchmark.txt permissions issue, right?

PeterSprague, 2017-10-18:
Nope, though I had to wrestle with Docker to get ODM to access the images with Fedora host.  I am now using an image I built on Sept/10/2017 from master.

timisonyourside, 2017-10-19:
Ok, may have found the answer in User Namespaces in the Docker Engine: https://success.docker.com/KBase/Introduction_to_User_Namespaces_in_Docker_Engine. 

Followed the instructions there to: 
1) activate my specific user namespace (mu 'user1' userid), 
2) Use my specific user namespace by stopping my docker instance and restating the docker daemon with the command 'sudo dockerd --userns-remap=user1 &', 
3) This creates a separate environment in /var/lib/docker for my specific user namespace - none of the root containers or images are usable in the new namespace, so 
4) I first removed all the directories that I had manually created by hand so that docker could recreate them under the new namespace and I removed the -u mapping, the -v benchmark mapping and the mappings for /etc/passwd and /etc/group  from the command as well and 
5) I reran my updated docker command which pulled a new copy of the OpenDroneMap image automatically as part of the docker run command for this new namespace. So far so good.

Here is an updated version or the new docker run command: docker run -v/home/user1/dev/sitemon/temp/a642f9060c37034f871ad5411ae495f2/original:/code/original -v/home/user1/dev/sitemon/temp/a642f9060c37034f871ad5411ae495f2/images:/code/images -v/home/user1/dev/sitemon/temp/a642f9060c37034f871ad5411ae495f2/opensfm:/code/opensfm -v/home/user1/dev/sitemon/temp/a642f9060c37034f871ad5411ae495f2/pmvs:/code/pmvs -v/home/user1/dev/sitemon/temp/a642f9060c37034f871ad5411ae495f2/odm_orthophoto:/code/odm_orthophoto -v/home/user1/dev/sitemon/temp/a642f9060c37034f871ad5411ae495f2/odm_georeferencing:/code/odm_georeferencing -v/home/user1/dev/sitemon/temp/a642f9060c37034f871ad5411ae495f2/odm_texturing:/code/odm_texturing -v/home/user1/dev/sitemon/temp/a642f9060c37034f871ad5411ae495f2/images_resize:/code/images_resize -v/home/user1/dev/sitemon/temp/a642f9060c37034f871ad5411ae495f2/odm_meshing:/code/odm_meshing -v/home/user1/dev/sitemon/temp/a642f9060c37034f871ad5411ae495f2/odm_mesh:/code/odm_mesh --rm -t opendronemap/opendronemap --force-ccd 17.3 --opensfm-processes 2 --resize-to 3200 --min-num-features 10000 --pmvs-min-images 5 --mesh-size 850000 --mesh-octree-depth 10 --time --end-with odm_georeferencing

Mapped directories for the docker volumes are now created by docker on the local host with the subuid as described in the article I followed and OpenDroneMap appears to be running without issue before opensfm.  I'll let you know once the job gets to opensfm if the permission problems are solved with this approach.

If it works, might be a recommendation to follow to get the permissions right from the start without having to make any changes to the OpenDroneMap Dockerfile. Basically just a change to the way the Dover daemon is started.

dakotabenjamin, 2018-03-28:
This issue was moved to http://community.opendronemap.org/t/odm-docker-file-code-opendm-system-py-line-53-errno-13-permission-denied-code-benchmark-txt/385

closed 2018-03-28 by dakotabenjamin