OpenDroneMap/ODM #687: Building nvidia/cuda:8.0-devel-ubuntu16.04 for GPU processing tests produces catkin error

PeterSprague, 2017-10-18:
Still pushing through this cuda-docker-opencv-2.4.13 build. Getting a catkin error on make:
Traceback (most recent call last):
File "/usr/local/src/OpenDroneMap/SuperBuild/build/ecto/catkin_generated/generate_cached_setup.py", line 20, in <module>
from catkin.environment_cache import generate_environment_script
ImportError: No module named catkin.environment_cache
CMake Error at /usr/local/src/OpenDroneMap/SuperBuild/install/share/catkin/cmake/safe_execute_process.cmake:11 (message):
execute_process(/usr/bin/python
"/usr/local/src/OpenDroneMap/SuperBuild/build/ecto/catkin_generated/generate_cached_setup.py")
returned error code 1

Using catkin 0.7.7
No cmake errors
Any ideas?

PeterSprague, 2017-10-18:
Detweaked all the cmake files, all back to current "master", build fails at etco:
-- downloading... done
-- verifying file...
file='/usr/local/src/OpenDroneMap/SuperBuild/download/ecto/c6178ed0102a66cebf503a4213c27b0f60cfca69.zip'
-- verifying file... done
-- extracting...
src='/usr/local/src/OpenDroneMap/SuperBuild/download/ecto/c6178ed0102a66cebf503a4213c27b0f60cfca69.zip'
dst='/usr/local/src/OpenDroneMap/SuperBuild/src/ecto'
-- extracting... [tar xfz]
-- extracting... [analysis]
-- extracting... [rename]
-- extracting... [clean up]
-- extracting... done
[ 32%] No patch step for 'ecto'
[ 33%] No update step for 'ecto'
[ 34%] Performing configure step for 'ecto'
-- The C compiler identification is GNU 5.4.0
-- The CXX compiler identification is GNU 5.4.0
-- Check for working C compiler: /usr/bin/cc
-- Check for working C compiler: /usr/bin/cc -- works
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Detecting C compile features
-- Detecting C compile features - done
-- Check for working CXX compiler: /usr/bin/c++
-- Check for working CXX compiler: /usr/bin/c++ -- works
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Using CATKIN_DEVEL_PREFIX: /usr/local/src/OpenDroneMap/SuperBuild/build/ecto/devel
-- Using CMAKE_PREFIX_PATH:
-- Found PythonInterp: /usr/bin/python (found version "2.7.12")
-- Using PYTHON_EXECUTABLE: /usr/bin/python
-- Using Debian Python package layout
-- Using empy: /usr/bin/empy
-- Using CATKIN_ENABLE_TESTING: OFF
-- catkin 0.6.16
Traceback (most recent call last):
File "/usr/local/src/OpenDroneMap/SuperBuild/build/ecto/catkin_generated/generate_cached_setup.py", line 20, in <module>
from catkin.environment_cache import generate_environment_script
ImportError: No module named catkin.environment_cache
CMake Error at /usr/local/src/OpenDroneMap/SuperBuild/install/share/catkin/cmake/safe_execute_process.cmake:11 (message):
execute_process(/usr/bin/python
"/usr/local/src/OpenDroneMap/SuperBuild/build/ecto/catkin_generated/generate_cached_setup.py")
returned error code 1
Call Stack (most recent call first):
/usr/local/src/OpenDroneMap/SuperBuild/install/share/catkin/cmake/all.cmake:186 (safe_execute_process)
/usr/local/src/OpenDroneMap/SuperBuild/install/share/catkin/cmake/catkinConfig.cmake:20 (include)
CMakeLists.txt:32 (find_package)

-- Configuring incomplete, errors occurred!

OpenDroneMap/SuperBuild/build/ecto/CMakeFiles/CMakeOutput.log doesn't show an error that I can see

Initial build image is: nvidia/cuda:8.0-devel-ubuntu16.04

pierotofy, 2017-10-18:
Mm, the only thing that pops out is this line:

`-- catkin 0.6.16`

Whereas you mentioned you are using 0.7.7. Perhaps it's a version mismatch issue? What happens if you downgrade to 0.6.16?

PeterSprague, 2017-10-18:
0.6.16 was second run using the original master CMake files

closed 2017-10-19 by PeterSprague