OpenDroneMap/ODM #688: What vertical datum is my data in?

gerry1138, 2017-10-18:
I have processed drone imagery using OpenDroneMap captured with a DJIP3P using the latest firmware.
The data were processed without any GCP data.  The drone is suppose to record elevation values in height above the home point.  The imagery was captured with the drone at 40m.  My resulting LAS points plot values between -2 and -13.  The elevation at this location is ~10feet(NAVD88).  What are the elevation values relative to?

PeterSprague, 2017-10-18:
This question belongs in ODM Gitter help or the new discourse discussion when on-line.  It is not a development issue.  Please move question to Gitter for help from a great group of users and developers.

To possibly answer your question, I believe all GPS use WGS-84 as their datum.  Not sure if that is really the issue - ~10ft would be ~3m, with GPS accuracy being 2-3m horizontal and 2-3X that vertical. My GPS logs show absolute height above sea level.

gerry1138, 2017-10-18:
Sorry for choosing the wrong venue but maybe it is an issue because the EXIF headers of the images show an elevation 114ft but the LAS files have z values of -2.114 to -13.199(meters ?).

PeterSprague, 2017-10-18:
No worries, was just a suggestion because often the Gitter help yields quicker response.  Might actually be an issue.  

If using altimeter, not GPS, then my response is not relevant.  Have revised my response.

dakotabenjamin, 2017-10-20:
Unless I'm mistaken, I believe we've resolved this issue.

closed 2017-10-20 by dakotabenjamin

gerry1138, 2017-10-23:
Yes. Resolved. The odd readings come from DroneDeploy which is basically making up elevation data. OpenDroneMap is returning the correct elevation values based on EXIF.
Open source wins again.
Thanks. Do I need to take any further action?

<div>-------- Original message --------</div><div>From: Dakota Benjamin <notifications@github.com> </div><div>Date:10/20/2017  4:26 PM  (GMT-08:00) </div><div>To: OpenDroneMap/OpenDroneMap <OpenDroneMap@noreply.github.com> </div><div>Cc: Gerry Gabrisch <gerry@gabrisch.us>,Author <author@noreply.github.com> </div><div>Subject: Re: [OpenDroneMap/OpenDroneMap] What vertical datum is my data in?  (#688) </div><div>
</div>Unless I'm mistaken, I believe we've resolved this issue.