OpenDroneMap/ODM #689: ODM Returns negative values on LAS

gerry1138, 2017-10-18:
I processed the images with ODM (link below) captured with a DJIP3P.  The DJIP3P has the latest firmware and is using the altimeter to record the height above ground.  The EXIF headers of the images show 117 ft altitude.  The DJI settings are set to capture feet.  The resulting LAS files has negative elevation values.   A processing of the same  imagery using DroneDeploy returned the correct LAS values.
https://www.dropbox.com/s/n3j7swg15pkire7/images.zip?dl=0

pierotofy, 2017-10-19:
Just to confirm, the `odm_georeferenced_model.las` file has negative elevation values?

gerry1138, 2017-10-19:
Yes. That is correct.

<div>-------- Original message --------</div><div>From: Piero Toffanin <notifications@github.com> </div><div>Date:10/19/2017  12:32 PM  (GMT-08:00) </div><div>To: OpenDroneMap/OpenDroneMap <OpenDroneMap@noreply.github.com> </div><div>Cc: Gerry Gabrisch <gerry@gabrisch.us>,Author <author@noreply.github.com> </div><div>Subject: Re: [OpenDroneMap/OpenDroneMap] ODM Returns negative values on LAS  (#689) </div><div>
</div>Just to confirm, the odm_georeferenced_model.las file has negative elevation values?

pierotofy, 2017-10-20:
The EXIF tags within the images seem to support the values outputted by ODM:

`# exiftool DJI_0017.JPG`

```
GPS Altitude                    : 35.2 m Below Sea Level
GPS Latitude                    : 48 deg 46' 46.46" N
GPS Longitude                   : 122 deg 35' 29.82" W
GPS Position                    : 48 deg 46' 46.46" N, 122 deg 35' 29.82" W
```
35.2 m `Below` Sea Level.

Not sure why the images are tagged as such, but ODM uses whatever the GPS Altitude value says, thus the negative elevation values.

pierotofy, 2017-10-20:
How are you checking the values for the EXIF tags?

gerry1138, 2017-10-20:
A Python script I wrote to read EXIF and generate KML file points for the image locations.

<div>-------- Original message --------</div><div>From: Piero Toffanin <notifications@github.com> </div><div>Date:10/20/2017  8:11 AM  (GMT-08:00) </div><div>To: OpenDroneMap/OpenDroneMap <OpenDroneMap@noreply.github.com> </div><div>Cc: Gerry Gabrisch <gerry@gabrisch.us>,Author <author@noreply.github.com> </div><div>Subject: Re: [OpenDroneMap/OpenDroneMap] ODM Returns negative values on LAS  (#689) </div><div>
</div>How are you checking the values for the EXIF tags?

gerry1138, 2017-10-20:
Wait...pyexiv2 library returns
Tag =Exif.GPSInfo.GPSAltitude
35233/1000

I do not see a negative value there.

<div>-------- Original message --------</div><div>From: Piero Toffanin <notifications@github.com> </div><div>Date:10/20/2017  8:10 AM  (GMT-08:00) </div><div>To: OpenDroneMap/OpenDroneMap <OpenDroneMap@noreply.github.com> </div><div>Cc: Gerry Gabrisch <gerry@gabrisch.us>,Author <author@noreply.github.com> </div><div>Subject: Re: [OpenDroneMap/OpenDroneMap] ODM Returns negative values on LAS  (#689) </div><div>
</div>The EXIF tags within the images seem to support the values outputted by ODM:

# exiftool DJI_0017.JPG

GPS Altitude                    : 35.2 m Below Sea Level
GPS Latitude                    : 48 deg 46' 46.46" N
GPS Longitude                   : 122 deg 35' 29.82" W
GPS Position                    : 48 deg 46' 46.46" N, 122 deg 35' 29.82" W
35.2 m Below Sea Level.

Not sure why the images are tagged as such, but ODM uses whatever the GPS Altitude value says, thus the negative elevation values.

dakotabenjamin, 2017-10-20:
There's a tag called AltutudeRef that will give the positive/negative value. Can you check that?

gerry1138, 2017-10-20:
GPSAltitudeRef returns 1

<div>-------- Original message --------</div><div>From: Dakota Benjamin <notifications@github.com> </div><div>Date:10/20/2017  12:09 PM  (GMT-08:00) </div><div>To: OpenDroneMap/OpenDroneMap <OpenDroneMap@noreply.github.com> </div><div>Cc: Gerry Gabrisch <gerry@gabrisch.us>,Author <author@noreply.github.com> </div><div>Subject: Re: [OpenDroneMap/OpenDroneMap] ODM Returns negative values on LAS  (#689) </div><div>
</div>There's a tag called AltutudeRef that will give the positive/negative value. Can you check that?

pierotofy, 2017-10-20:
https://www.awaresystems.be/imaging/tiff/tifftags/privateifd/gps/gpsaltituderef.html

```
If the reference is sea level and the altitude is above sea level, 0 is given. If the altitude is below sea level, a value of 1 is given and the altitude is indicated as an absolute value in the GPSAltitude tag. The reference unit is meters. Note that this tag is BYTE type, unlike other reference tags.

The specification defines these values:

0 = Above sea level
1 = Below sea level
```

gerry1138, 2017-10-20:
Yes. Thanks. I found that online. I did not know. I guess DroneDeploy ignores the referece value. Shame on them!

Thanks for working on ODM. I appreciate your hard work and help.

<div>-------- Original message --------</div><div>From: Piero Toffanin <notifications@github.com> </div><div>Date:10/20/2017  1:13 PM  (GMT-08:00) </div><div>To: OpenDroneMap/OpenDroneMap <OpenDroneMap@noreply.github.com> </div><div>Cc: Gerry Gabrisch <gerry@gabrisch.us>,Author <author@noreply.github.com> </div><div>Subject: Re: [OpenDroneMap/OpenDroneMap] ODM Returns negative values on LAS  (#689) </div><div>
</div>https://www.awaresystems.be/imaging/tiff/tifftags/privateifd/gps/gpsaltituderef.html

If the reference is sea level and the altitude is above sea level, 0 is given. If the altitude is below sea level, a value of 1 is given and the altitude is indicated as an absolute value in the GPSAltitude tag. The reference unit is meters. Note that this tag is BYTE type, unlike other reference tags.

The specification defines these values:

0 = Above sea level
1 = Below sea level

dakotabenjamin, 2017-10-20:
It looks like this problem is solved as it was determined to be a data problem. Please reopen if you have further issues.

closed 2017-10-20 by dakotabenjamin