OpenDroneMap/ODM #691: "--resize-to -1" doesn't work

tmizu23, 2017-10-20:
if "--resize-to" is -1, odm_georeferencing.py or Georef.cpp does't process as original size. 
value -1 valid only opensfm process.

dakotabenjamin, 2017-10-20:
Hi,
Are you running from master, or v0.3.1?

tmizu23, 2017-10-20:
I run ti by windows docker latest.

dakotabenjamin, 2017-10-20:
Instead of running `--resize-to -1` try `--skip-resize`

tmizu23, 2017-10-21:
Hi,thank you reply.
--skip-resize was removed at "Remove Resize module" committed on 25 Aug

tmizu23, 2017-10-21:
I tried to fix this issue by changing Georef.cpp. 

and this code will also fix the issue that gcp with resized image by --resize-to doesn't return correct result. When using GCP, the previous code seems to be the correct result only for the original size image.

and I think that gcp_list.txt should be made with the original image size even if setting the resize option, so I changed the code like that.  ref. #541 #632 

Please check the code.
Changing Georef.cpp from L797   https://github.com/OpenDroneMap/OpenDroneMap/blob/master/modules/odm_georef/src/Georef.cpp#L797
    
        // Caluculate resize scale
        double resize_scale = 1.0;
        cv::Mat image = cv::imread(cam.texture_file);
        if(bundleResizedTo_ == -1) bundleResizedTo_ = image.cols;
        resize_scale = static_cast<double>(bundleResizedTo_)/image.cols;
        
        // Rescale the heigh, width, focal_length 
        cam.height = static_cast<double>(image.rows) * resize_scale;
        cam.width = static_cast<double>(image.cols) * resize_scale;
        cam.focal_length *= resize_scale;
        
        // Rescale the pixcelX, pixcelY of gcp_list.txt
        // The pixel position for the GCP in pcl-format in order to use pcl-functions 
        pcl::PointXY gcpPos;
        gcpPos.x = static_cast<float>(gcps_[gcpIndex].pixelX_ * resize_scale );
        gcpPos.y = static_cast<float>(gcps_[gcpIndex].pixelY_ * resize_scale);

dakotabenjamin, 2017-10-22:
Sorry, I forgot that was my removal 😨 

So now in the code there is no resizing done, the --resize-to only applies to OpenSfM for speeding up the process. Everything else uses the original images for maximum resolution. When `georef` is called, it uses the original image size, therefore resize_scale should always be 1. We should simply remove all the resize stuff from odm_georeferencing to make it simple.

tmizu23, 2017-10-23:
almost +1,
but there is a bug at the moment, so please fix following points.

Georef.cpp L800 should be commented out for georeferencing correctly
https://github.com/OpenDroneMap/OpenDroneMap/blob/master/modules/odm_georef/src/Georef.cpp#L800

dakotabenjamin, 2017-11-01:
This should be fixed in latest master

closed 2017-11-01 by dakotabenjamin