OpenDroneMap/ODM #694: OpenCV-3.3.0 w/ nvidia/cuda:8.0-devel-ubuntu16.04 image for GPU processing tests

PeterSprague, 2017-10-20:
OpenCV-2.4.xx w/ Cuda does not recognize newer classes of NVidia GPU when building, aka only up to Kepler.  I need Pascal, therefore need to build with OpenCV-3.3.0

Updated SuperBuild/CMakeLists.txt & Extenal-xxxx.cmake to newest versions:
Ecto --> 0.6.12
OpenCV-2.4.11 --> 3.3.0 
PCL 1.8.0 -->1.8.1  

On host:
$ docker build --tag gei/opendronemap:cuda-8.0-opencv-3.3.0
Builds successfully through [rest of Dockerfile is commented out]:
$ cd SuperBuild/build $$ cmake .. $$ make -j20  no errors
$ docker run -it --name odm-cuda-8.0-opencv-3.3.0 gei/opendronemap:cuda-8.0-opencv-3.3.0 bash
In container -> moving up to next Dockerfile step:
$ cd ../.. && mkdir build && cd build && cmake .. 
a few warnings, but
-- Configuring done
-- Generating done
-- Build files have been written to: /code/build

$ make -j20 throws errors at odm_texturing:
[ 86%] Linking CXX executable ../../bin/odm_meshing
[ 86%] Built target odm_meshing
[ 89%] Linking CXX executable ../../bin/odm_texturing
CMakeFiles/odm_texturing.dir/src/OdmTexturing.cpp.o: In function `OdmTexturing::loadCameras()':
OdmTexturing.cpp:(.text+0x2a3b): undefined reference to `cv::imread(cv::String const&, int)'
CMakeFiles/odm_texturing.dir/src/OdmTexturing.cpp.o: In function `OdmTexturing::createTextures()':
OdmTexturing.cpp:(.text+0x84ef): undefined reference to `cv::imread(cv::String const&, int)'
OdmTexturing.cpp:(.text+0x8a43): undefined reference to `cv::imwrite(cv::String const&, cv::_InputArray const&, std::vector<int, std::allocator<int> > const&)'
OdmTexturing.cpp:(.text+0x8d35): undefined reference to `cv::imwrite(cv::String const&, cv::_InputArray const&, std::vector<int, std::allocator<int> > const&)'
collect2: error: ld returned 1 exit status
modules/odm_texturing/CMakeFiles/odm_texturing.dir/build.make:479: recipe for target 'bin/odm_texturing' failed
make[2]: *** [bin/odm_texturing] Error 1
CMakeFiles/Makefile2:338: recipe for target 'modules/odm_texturing/CMakeFiles/odm_texturing.dir/all' failed
make[1]: *** [modules/odm_texturing/CMakeFiles/odm_texturing.dir/all] Error 2
make[1]: *** Waiting for unfinished jobs....
[ 93%] Linking CXX executable ../../bin/odm_orthophoto
CMakeFiles/odm_orthophoto.dir/src/OdmOrthoPhoto.cpp.o: In function `OdmOrthoPhoto::createOrthoPhoto()':
OdmOrthoPhoto.cpp:(.text+0x916c): undefined reference to `cv::imread(cv::String const&, int)'
OdmOrthoPhoto.cpp:(.text+0x96bb): undefined reference to `cv::imwrite(cv::String const&, cv::_InputArray const&, std::vector<int, std::allocator<int> > const&)'
collect2: error: ld returned 1 exit status
modules/odm_orthophoto/CMakeFiles/odm_orthophoto.dir/build.make:453: recipe for target 'bin/odm_orthophoto' failed
make[2]: *** [bin/odm_orthophoto] Error 1
CMakeFiles/Makefile2:283: recipe for target 'modules/odm_orthophoto/CMakeFiles/odm_orthophoto.dir/all' failed
make[1]: *** [modules/odm_orthophoto/CMakeFiles/odm_orthophoto.dir/all] Error 2
[ 96%] Linking CXX executable ../../bin/odm_25dmeshing
[ 96%] Built target odm_25dmeshing
[100%] Linking CXX executable ../../bin/odm_georef
CMakeFiles/odm_georef.dir/src/Georef.cpp.o: In function `Georef::performGeoreferencingWithGCP()':
Georef.cpp:(.text+0xaec2): undefined reference to `cv::imread(cv::String const&, int)'
collect2: error: ld returned 1 exit status
modules/odm_georef/CMakeFiles/odm_georef.dir/build.make:506: recipe for target 'bin/odm_georef' failed
make[2]: *** [bin/odm_georef] Error 1
CMakeFiles/Makefile2:173: recipe for target 'modules/odm_georef/CMakeFiles/odm_georef.dir/all' failed
make[1]: *** [modules/odm_georef/CMakeFiles/odm_georef.dir/all] Error 2
Makefile:83: recipe for target 'all' failed
make: *** [all] Error 2

Ideas?  I'm thinking that possibly the ODM scripts calls need to be tuned for the newer versions of software?

dakotabenjamin, 2017-10-20:
You're right. Checkout the changes in my 16.10 pull request #677 those errors look familiar and I think I fixed them in that.

PeterSprague, 2017-10-20:
Thanks, included your commits into my cuda branch.  Performing a docker build now.  Will report on the outcome one way or another.

PeterSprague, 2017-10-21:
Fails at the same point with same errors.  That would have been too easy.

Time for a beer, and leave this for a bit

dakotabenjamin, 2017-10-22:
Oh, try deleting the build commands for odm_texturing. That's way outdated and deprecated code.

PeterSprague, 2017-10-22:
Hmm, but I'm working off an upstream merge of the master fork?

dakotabenjamin, 2017-10-22:
Yeah, it hasnt been removed in master yet... whoops

PeterSprague, 2017-10-22:
If I'm understanding, remove the statement:
RUN cd ../.. && mkdir build && cd build && cmake .. && make -j$(nproc)

aka don't do the second build outside SuperBuild?

dakotabenjamin, 2017-10-22:
Delete line 6 here: https://github.com/OpenDroneMap/OpenDroneMap/blob/master/modules/CMakeLists.txt#L6

PeterSprague, 2017-10-22:
add_subdirectory(odm_texturing)?

dakotabenjamin, 2017-10-22:
yeah

PeterSprague, 2017-10-22:
Thanks, doing docker build on revised code.

PeterSprague, 2017-10-22:
modules/odm_meshing/CMakeFiles/odm_meshing.dir/build.make:62: recipe for target 'modules/odm_meshing/CMakeFiles/odm_meshing.dir/src/OdmMeshing.cpp.o' failed
make[2]: *** [modules/odm_meshing/CMakeFiles/odm_meshing.dir/src/OdmMeshing.cpp.o] Error 1
make[2]: *** Waiting for unfinished jobs....
CMakeFiles/Makefile2:225: recipe for target 'modules/odm_meshing/CMakeFiles/odm_meshing.dir/all' failed
make[1]: *** [modules/odm_meshing/CMakeFiles/odm_meshing.dir/all] Error 2
make[1]: *** Waiting for unfinished jobs....
[ 87%] Linking CXX executable ../../bin/odm_orthophoto
CMakeFiles/odm_orthophoto.dir/src/OdmOrthoPhoto.cpp.o: In function `OdmOrthoPhoto::createOrthoPhoto()':
OdmOrthoPhoto.cpp:(.text+0x916c): undefined reference to `cv::imread(cv::String const&, int)'
OdmOrthoPhoto.cpp:(.text+0x96bb): undefined reference to `cv::imwrite(cv::String const&, cv::_InputArray const&, std::vector<int, std::allocator<int> > const&)'
collect2: error: ld returned 1 exit status
modules/odm_orthophoto/CMakeFiles/odm_orthophoto.dir/build.make:453: recipe for target 'bin/odm_orthophoto' failed
make[2]: *** [bin/odm_orthophoto] Error 1
CMakeFiles/Makefile2:280: recipe for target 'modules/odm_orthophoto/CMakeFiles/odm_orthophoto.dir/all' failed
make[1]: *** [modules/odm_orthophoto/CMakeFiles/odm_orthophoto.dir/all] Error 2
[ 91%] Linking CXX executable ../../bin/odm_25dmeshing
[ 91%] Built target odm_25dmeshing
[ 95%] Linking CXX executable ../../bin/odm_georef
CMakeFiles/odm_georef.dir/src/Georef.cpp.o: In function `Georef::performGeoreferencingWithGCP()':
Georef.cpp:(.text+0xaec2): undefined reference to `cv::imread(cv::String const&, int)'
collect2: error: ld returned 1 exit status
modules/odm_georef/CMakeFiles/odm_georef.dir/build.make:506: recipe for target 'bin/odm_georef' failed
make[2]: *** [bin/odm_georef] Error 1
CMakeFiles/Makefile2:170: recipe for target 'modules/odm_georef/CMakeFiles/odm_georef.dir/all' failed
make[1]: *** [modules/odm_georef/CMakeFiles/odm_georef.dir/all] Error 2
Makefile:83: recipe for target 'all' failed
make: *** [all] Error 2

dakotabenjamin, 2017-10-22:
It looks like it's not finding OpenCV. In odm_orthophoto for example, cmake looks for OpenCV here: https://github.com/OpenDroneMap/OpenDroneMap/blob/master/modules/odm_orthophoto/CMakeLists.txt#L15

I'm at a conference right now or I'd be able to diagnose the issue further. Are you installing OpenCV inside the SuperBuild? 

You may also want to look at: https://docs.opencv.org/3.3.0/d4/da8/group__imgcodecs.html#gabbc7ef1aa2edfaa87772f1202d67e0ce

PeterSprague, 2017-10-22:
Weird, pretty sure its building OpenCV without errors.  Using docker build to build the image up to the point of failure at ../.. cmake .. && make. Will poke about after another fresh run.

Building a gist of my steps and the cuda-8.0-opencv-3.3.0 branch is pushed to my fork @https://github.com/GeoVisionEnvironmental/OpenDroneMap.git

Enjoy your conference, heading out for a trail run.  Back at this later

PeterSprague, 2017-10-24:
Update all the modules/xx/CMakeLists.txt PCL 1.8 --> 1.8.1

SuperBuild cmake and make builds without errors. 
/code/build cmake shows finds OpenCV & PCL.
/code/build make -j20 errors still:

modules/odm_meshing/CMakeFiles/odm_meshing.dir/build.make:62: recipe for target 'modules/odm_meshing/CMakeFiles/odm_meshing.dir/src/OdmMeshing.cpp.o' failed
make[2]: *** [modules/odm_meshing/CMakeFiles/odm_meshing.dir/src/OdmMeshing.cpp.o] Error 1
make[2]: *** Waiting for unfinished jobs....
CMakeFiles/Makefile2:225: recipe for target 'modules/odm_meshing/CMakeFiles/odm_meshing.dir/all' failed
make[1]: *** [modules/odm_meshing/CMakeFiles/odm_meshing.dir/all] Error 2
make[1]: *** Waiting for unfinished jobs....
[ 87%] Linking CXX executable ../../bin/odm_orthophoto
CMakeFiles/odm_orthophoto.dir/src/OdmOrthoPhoto.cpp.o: In function `OdmOrthoPhoto::createOrthoPhoto()':
OdmOrthoPhoto.cpp:(.text+0x916c): undefined reference to `cv::imread(cv::String const&, int)'
OdmOrthoPhoto.cpp:(.text+0x96bb): undefined reference to `cv::imwrite(cv::String const&, cv::_InputArray const&, std::vector<int, std::allocator<int> > const&)'
collect2: error: ld returned 1 exit status
modules/odm_orthophoto/CMakeFiles/odm_orthophoto.dir/build.make:453: recipe for target 'bin/odm_orthophoto' failed
make[2]: *** [bin/odm_orthophoto] Error 1
CMakeFiles/Makefile2:280: recipe for target 'modules/odm_orthophoto/CMakeFiles/odm_orthophoto.dir/all' failed
make[1]: *** [modules/odm_orthophoto/CMakeFiles/odm_orthophoto.dir/all] Error 2
[ 91%] Linking CXX executable ../../bin/odm_25dmeshing
[ 91%] Built target odm_25dmeshing
[ 95%] Linking CXX executable ../../bin/odm_georef
CMakeFiles/odm_georef.dir/src/Georef.cpp.o: In function `Georef::performGeoreferencingWithGCP()':
Georef.cpp:(.text+0xaec2): undefined reference to `cv::imread(cv::String const&, int)'
collect2: error: ld returned 1 exit status
modules/odm_georef/CMakeFiles/odm_georef.dir/build.make:506: recipe for target 'bin/odm_georef' failed
make[2]: *** [bin/odm_georef] Error 1
CMakeFiles/Makefile2:170: recipe for target 'modules/odm_georef/CMakeFiles/odm_georef.dir/all' failed
make[1]: *** [modules/odm_georef/CMakeFiles/odm_georef.dir/all] Error 2
Makefile:83: recipe for target 'all' failed
make: *** [all] Error 2

Will keep poking about

PeterSprague, 2017-10-26:
Update:
Just posted a Gist with my build steps and usage: https://gist.github.com/PeterSprague/98cc0ae744e8285111978167e5626678

Pushed latest to: https://github.com/GeoVisionEnvironmental/OpenDroneMap.git  --> branch cuda-8.0-opencv-3.3.1

Builds w/o errors, but when run against a know set of images throws a python error:
$ nvidia-docker run -it --name odm-cuda-8.0-opencv-3.3.0 -v $(pwd)/images/images_jpg:/code/images:Z -v $(pwd)/images_resize:/code/images_resize:Z -v $(pwd)/opensfm:/code/opensfm:Z -v $(pwd)/odm_meshinging:/code/odm_meshing:Z -v $(pwd)/odm_texturing:/code/odm_texturing:Z -v $(pwd)/odm_georeferencing:/code/odm_georeferencing:Z -v $(pwd)/odm_orthophoto:/code/odm_orthophoto:Z -v $(pwd)/settings.yaml:/code/settings.yaml:Z --rm gei/opendronemap:cuda-8.0-opencv-3.3.0  --force-focal 3.97 --force-ccd 4.63104
Error:
~
2017-10-26 04:04:25,915 Found 18564 points in 11.1263718605s
2017-10-26 04:04:26,052 Found 18469 points in 11.6344058514s
2017-10-26 04:04:27,398 Extracting ROOT_HAHOG features for image 2017_0908_135243_192.JPG
2017-10-26 04:04:27,398 Extracting ROOT_HAHOG features for image 2017_0908_134607_010.JPG
2017-10-26 04:04:27,399 Extracting ROOT_HAHOG features for image 2017_0908_135410_234.JPG
Traceback (most recent call last):
  File "/code/SuperBuild/src/opensfm/bin/opensfm", line 34, in <module>
    command.run(args)
  File "/code/SuperBuild/src/opensfm/opensfm/commands/detect_features.py", line 33, in run
    p.map(detect, arguments)
  File "/usr/lib/python2.7/multiprocessing/pool.py", line 251, in map
    return self.map_async(func, iterable, chunksize).get()
  File "/usr/lib/python2.7/multiprocessing/pool.py", line 567, in get
    raise self._value
AttributeError: 'module' object has no attribute 'Index'
Traceback (most recent call last):
  File "/code/run.py", line 47, in <module>
    plasm.execute(niter=1)
  File "/code/scripts/opensfm.py", line 97, in process
    (context.pyopencv_path, context.opensfm_path, tree.opensfm))
  File "/code/opendm/system.py", line 34, in run
    raise Exception("Child returned {}".format(retcode))
Exception: Child returned 1

dakotabenjamin, 2017-10-26:
You're not the only one: https://github.com/mapillary/OpenSfM/issues/228

PeterSprague, 2017-10-26:
Interesting, will have to let sit for now, see what comes back for solutions.

ODM master hasn't updated its OpenSFM version recently?

dakotabenjamin, 2017-10-26:
#699 would do that. It looks like you're building 0.1.0, right? Current master is [from July](https://github.com/mapillary/OpenSfM/commit/0a5a6f07b66088e9b63bddaf29bb8f49dd39f9e4) which is actually slightly newer than 0.1.0

PeterSprague, 2017-10-26:
I pulled in latest upstream master to then branch and build ODM-Cuda.

dakotabenjamin, 2017-10-26:
Ok try it now, we just updated OpenSfM.

PeterSprague, 2017-10-26:
Merged #699 loky fix, building Docker image now.

PeterSprague, 2017-10-26:
Is this required for ODM?  I had installed for testing, but not for last build of ODM

OpenSfM Install note:

When running OpenSfM on top of OpenCV 3.0 the OpenCV Contrib modules are required for extracting SIFT or SURF features.

dakotabenjamin, 2017-10-26:
No, we use HAHOG not surf or sift

dakotabenjamin, 2017-10-26:
Does this dataset work with standard ODM? Have you tried on a different dataset that is known to work othewise?

PeterSprague, 2017-10-26:
Yes, the dataset works with my September ODM docker build.  Why I choose it, but can try another to test if difference in errors

dakotabenjamin, 2017-10-26:
Yeah I don't know what else to do at this point

PeterSprague, 2017-10-26:
I will pull and build a new docker image from the latest master.  See if that runs.

PeterSprague, 2017-10-27:
- gei-sept102017 build from September ODM master --> completes no errors and produces ortho
- odm-master built from today's commits with loky --> completes no errors and produces ortho
- gei-cuda-8.0-opencv-3.3.1 built from today's commits with loky --> continues to error as above

Would someone be able to check the diffs?  https://github.com/OpenDroneMap/OpenDroneMap/compare/master...GeoVisionEnvironmental:cuda-8.0-opencv-3.3.1

I can't see anything that should/could cause this.  Someone with better eyes/more knowledge may be able to find a solution.

dakotabenjamin, 2017-10-27:
So odd that you get the same error with two different parallelization methods, and it's not a data issue. 

The diff looks fine to me. Can you try running with `--opensfm-processes 1`

PeterSprague, 2017-10-27:
The two errors posted above are different.  Last one refers to "loky", earlier one refers to "multiprocessing".  Last run sustained much deeper into the process before erroring.

Ran with --opensfm=1:
2017-10-27 17:50:22,457 Extracting focal lengths for image 2017_0908_134743_054.JPG
2017-10-27 17:50:22,815 Extracting focal lengths for image 2017_0908_135448_254.JPG
[DEBUG]   running PYTHONPATH=/code/SuperBuild/install/lib/python2.7/dist-packages /code/SuperBuild/src/opensfm/bin/opensfm detect_features /code/opensfm
2017-10-27 17:50:23,533 Extracting ROOT_HAHOG features for image 2017_0908_134747_056.JPG
2017-10-27 17:50:31,038 Found 18515 points in 7.05036902428s
Traceback (most recent call last):
  File "/code/SuperBuild/src/opensfm/bin/opensfm", line 34, in <module>
    command.run(args)
  File "/code/SuperBuild/src/opensfm/opensfm/commands/detect_features.py", line 29, in run
    parallel_map(detect, arguments, processes)
  File "/code/SuperBuild/src/opensfm/opensfm/context.py", line 17, in parallel_map
    return map(func, args)
  File "/code/SuperBuild/src/opensfm/opensfm/commands/detect_features.py", line 63, in detect
    index = features.build_flann_index(f_sorted, data.config)
  File "/code/SuperBuild/src/opensfm/opensfm/features.py", line 299, in build_flann_index
    flann_Index = cv2.flann.Index if context.OPENCV3 else cv2.flann_Index
AttributeError: 'module' object has no attribute 'Index'
Traceback (most recent call last):
  File "/code/run.py", line 47, in <module>
    plasm.execute(niter=1)
  File "/code/scripts/opensfm.py", line 97, in process
    (context.pyopencv_path, context.opensfm_path, tree.opensfm))
  File "/code/opendm/system.py", line 34, in run
    raise Exception("Child returned {}".format(retcode))
Exception: Child returned 1

Different error than either above.

dakotabenjamin, 2017-10-27:
Yeah I think you're on to something, i'm proing over the cv2 docs right now...

Maybe flann wasn't installed correctly?

PeterSprague, 2017-10-27:
Would it make any sense to build a non-cuda version with opencv-3.3.1 and test?

PeterSprague, 2017-10-27:
gei-nocuda-opencv-3.3.1 borks.  Only thing changed from master was to revise the OpenCV version in SuperBuild/cmake/Ex...txt and CMakeLists.txt file

dakotabenjamin, 2017-10-31:
Just saw this: https://github.com/mapillary/OpenSfM/commit/b05ccbe75f74b16e1ad05eb1ae32f38188843a62

Can you pull latest opensfm and try again?

dakotabenjamin, 2017-11-02:
@PeterSprague did the above suggestion help?

PeterSprague, 2017-11-02:
Will try when I get back from Toronto.  Thanks for the pointer

PeterSprague, 2017-11-06:
What is the archive?  Don't see it listed in OpenSfM

PeterSprague, 2017-11-07:
Run with OpenSfM master with #231 commit that fixes flann opencv issue:

Runs most of the process, but errors out:
2017-11-07 18:49:44,697 Pruning depthmap for image 2017_0908_134614_014.JPG
2017-11-07 18:49:46,810 Pruning depthmap for image 2017_0908_135546_284.JPG
2017-11-07 18:49:49,513 Pruning depthmap for image 2017_0908_135654_316.JPG
2017-11-07 18:50:11,622 Merging depthmaps
[DEBUG]   running PYTHONPATH=/code/SuperBuild/install/lib/python2.7/dist-packages /code/SuperBuild/src/opensfm/bin/export_bundler /code/opensfm
[INFO]    Running ODM OpenSfM Cell - Finished
[INFO]    Running ODM Meshing Cell
[DEBUG]   Writing ODM Mesh file in: /code/odm_meshing/odm_mesh.ply
[DEBUG]   running /code/build/bin/odm_meshing -inputFile /code/opensfm/depthmaps/merged.ply -outputFile /code/odm_meshing/odm_mesh.ply -logFile /code/odm_meshing/odm_meshing_log.txt -maxVertexCount 30000 -octreeDepth 5 -verbose -samplesPerNode 1.0 -solverDivide 9
/code/build/bin/odm_meshing: error while loading shared libraries: libvtkCommonCore-6.2.so.6.2: cannot open shared object file: No such file or directory
Traceback (most recent call last):
  File "/code/run.py", line 47, in <module>
    plasm.execute(niter=1)
  File "/code/scripts/odm_meshing.py", line 82, in process
    '-samplesPerNode {samples} -solverDivide {solver}'.format(**kwargs))
  File "/code/opendm/system.py", line 34, in run
    raise Exception("Child returned {}".format(retcode))
Exception: Child returned 127

OpenSfM is not utilizing the GPU though:
$ nvidia-smi
Tue Nov  7 09:50:17 2017       
+-----------------------------------------------------------------------------+
| NVIDIA-SMI 384.90                 Driver Version: 384.90                    |
|-------------------------------+----------------------+----------------------+
| GPU  Name        Persistence-M| Bus-Id        Disp.A | Volatile Uncorr. ECC |
| Fan  Temp  Perf  Pwr:Usage/Cap|         Memory-Usage | GPU-Util  Compute M. |
|===============================+======================+======================|
|   0  GeForce GTX 1080    Off  | 00000000:05:00.0  On |                  N/A |
| 27%   34C    P8     8W / 180W |    451MiB /  8112MiB |      4%      Default |
+-------------------------------+----------------------+----------------------+
                                                                               
+-----------------------------------------------------------------------------+
| Processes:                                                       GPU Memory |
|  GPU       PID   Type   Process name                             Usage      |
|=============================================================================|
|    0      1784      G   /usr/libexec/Xorg                            414MiB |
|    0      2415      G   /usr/lib64/firefox/firefox                     3MiB |
|    0     20753      G   /proc/self/exe                                29MiB |
+-----------------------------------------------------------------------------+

I thought OpenSfM would pick up the GPU if OpenCV was built against Cuda?  Since this is not directly an ODM issue, I will continue this GPU issue over at OpenSfM.

closed 2017-11-08 by PeterSprague