OpenDroneMap/ODM #695: Can't find PCLConfig.cmake

yuboona, 2017-10-21:
I'm using ubuntu 16.04.And when I was trying to compile,I finally get the error:
   `CMake Error at modules/odm_orthophoto/CMakeLists.txt:12 (find_package):
  Could not find a package configuration file provided by "PCL" (requested
  version 1.7) with any of the following names:

    PCLConfig.cmake
    pcl-config.cmake

  Add the installation prefix of "PCL" to CMAKE_PREFIX_PATH or set "PCL_DIR"
  to a directory containing one of the above files.  If "PCL" provides a
  separate development package or SDK, be sure it has been installed.`
I have tried to find out answer in issues,but ailed,they usually get a 'Warning' about PCL ,while mine  is a 'Error'

dakotabenjamin, 2017-10-22:
Try running `./configure.sh install` again, sometimes the installation process runs into race condition issues and doing it again will fix.

dakotabenjamin, 2017-10-27:
any luck?

yuboona, 2017-12-27:
@dakotabenjamin Thank you very much.I tried reinstall ODM many times,but didn't work.Then I reinstalled my Ubuntu16.04, and it works!!! I think the failure was caused by my own fault that updated my Ubuntu16.04 too often .It may have caused my system's lib too new and can't satisfy the lib requirements of ODM.

yuboona, 2018-01-20:
I finally find the failure exactly is caused by race condition issue. So I modified the Cmake to make the downloads of 3D party software become the local source, Then everything was compiled really well.

dakotabenjamin, 2018-01-22:
Great! Do you mind sharing the changes you made?

closed 2018-01-22 by dakotabenjamin

yuboona, 2018-01-23:
sure，I‘m really happy to do that and it's just a so small change.

In the /SuperBuild/cmake  directory，I changed the .cmake files‘s content which are ：
```
#–Download step————–
DOWNLOAD_DIR ${SB_DOWNLOAD_DIR}
URL https://github.com/gflags/gflags/archive/v2.1.2.zip
URL_MD5 5cb0a1b38740ed596edb7f86cd5b3bd8
```
I changed them to：
```
#–Download step————–
DOWNLOAD_COMMAND “”
```

So now the autodownloads of cmake will not work. Then I just need to correctly extract the 3rd party source zip files to the directorys below /Superbuild/src （yes，we need to create some folders there）

finally，run the install of ODM，we can happily find it works.