OpenDroneMap/ODM #697: Problem building ODM image

stucraw, 2017-10-22:
I am trying to install OpenDroneMap with *Option 2: Installation from source*, as described here: https://github.com/OpenDroneMap/OpenDroneMap/wiki/Docker

I am running MacOS 10.12, using Docker.

If I run `docker images`, the my_odm_image appears to be partially created, but not completely:

REPOSITORY | TAG | IMAGE ID | CREATED | SIZE
------------ | ------------- | ------------- | ------------- | -------------
none | none | 3f98655413c0 | 12 minutes ago | 1.75GB

Below is the error when I run the build command:

> make: *** [all] Error 2
> Makefile:83: recipe for target 'all' failed
> The command '/bin/sh -c cd SuperBuild && mkdir build && cd build && cmake  .. && make -j$(nproc)     && cd ../.. && mkdir build && cd build && cmake .. && make -j$(nproc)' returned a non-zero code: 2

dakotabenjamin, 2017-10-22:
Try first deleting the image and rerunning the build, using `docker rmi 3f98655413c0`

You could also pull a prebuilt image rather than build from source using `docker pull opendronemap/opendronemap`

arkadyark, 2018-01-31:
I'm having this exact same error, failing at the Superbuild step with `Makefile:83: recipe for target 'all' failed`. I've tried it with deleting the image and rerunning, to no avail. Any suggestions?

smathermather, 2018-02-11:
I am going to close this as this was answered a while ago. Feel free to open it back up if you have more questions.

closed 2018-02-11 by smathermather