OpenDroneMap/ODM #698: Can't parse cordinate system at high lattitudes

Wes-Skeeter, 2017-10-25:
I have a set of images from UTM zone 8 taken at about 69 degrees N.   [Here](https://drive.google.com/file/d/0B7GsdLnhj_QfM2pucG5PbVNGTXc/view?usp=sharing) is a small subset of my dataset

Running:

docker run -it --rm -v $(pwd)/images:/code/images -v $(pwd)/odm_orthophoto:/code/odm_orthophoto -v $(pwd)/odm_georeferencing:/code/odm_georeferencing opendronemap/opendronemap --mesh-size 1000000 --force-ccd 8.8

throws this error:

Traceback (most recent call last): File "/code/run.py", line 47, in <module> plasm.execute(niter=1) File "/code/scripts/odm_georeferencing.py", line 164, in process geo_ref.parse_coordinate_system(tree.odm_georeferencing_coords) File "/code/opendm/types.py", line 317, in parse_coordinate_system self.utm_pole = ref[2][len(ref) - 1] IndexError: string index out of range

dakotabenjamin, 2017-10-27:
Have you run `docker pull opendronemap/opendronemap` recently?

Wes-Skeeter, 2017-10-27:
I had run `docker pull opendronemap/opendronemap` as recently a few days ago.  However, after encountering the problem, I gave up on trying to run through docker and installed it directly.  Looks like the version I installed is not the most up to date release, but I did manage to patch it well enough to work for my purposes. [#303](https://github.com/OpenDroneMap/WebODM/issues/303)

closed 2018-03-28 by dakotabenjamin