OpenDroneMap/ODM #700: odm generates a lot larger size .jpg files under /tmp

bingzhang, 2017-10-26:
Hello:

I am using odm docker image to process images, it seems to generate .jpg files under /tmp folder. e.g., IMG_160729_073053_0347_RGB.JPG is my input file which is 2.2 MB, and odm generates IMG_160729_073053_0347_RGB.JPG.jpg which is 24MB under /tmp folder.

May I know what are those .jpg files, and why the size is 10 times large compared with input file.

Is there a way to let odm to generate those .jpg files into tmp folder under project folder?

Regards.

dakotabenjamin, 2017-10-26:
I want to look into this, can you remind me how to go inside a running docker container? What's your run command?

bingzhang, 2017-10-26:
Yes,

You can use this command to go inside running container.
`docker exec -it cotainerid bash`

there is the example:
```
wifi-60-41:Downloads bing$ docker run -it --rm -v $(pwd)/images:/code/images opendronemap/opendronemap
wifi-60-41:Downloads bing$ docker ps | grep opendronemap
2f397f6fd717        opendronemap/opendronemap   "python /code/run...."   12 seconds ago      Up 12 seconds                                                                                                    reverent_rosalind
wifi-60-41:Downloads bing$ docker exec -it 2f397f6fd717 bash

```

watch the log of odm, when you saw this log, then you can see many .jpg files under /tmp folder.
Generating texture atlases:
	Sorting texture patches... done.
	Working on atlas 5 100%... done.
	Finalizing texture atlases... done. (Took: 6.353s)
Building objmodel:

root@2f397f6fd717:/tmp# ls -lart
total 359268
drwxr-xr-x 1 root root     4096 Oct 26 18:56 ..
-rw-r--r-- 1 root root 22522499 Oct 26 18:58 IMG_160729_073111_0351_RGB.JPG.jpg
-rw-r--r-- 1 root root 22550999 Oct 26 18:58 IMG_160729_073109_0350_RGB.JPG.jpg
-rw-r--r-- 1 root root 22418029 Oct 26 18:58 IMG_160729_073042_0342_RGB.JPG.jpg
-rw-r--r-- 1 root root 24795674 Oct 26 18:58 IMG_160729_073053_0347_RGB.JPG.jpg
-rw-r--r-- 1 root root 22740352 Oct 26 18:58 IMG_160729_073113_0352_RGB.JPG.jpg
-rw-r--r-- 1 root root 25599591 Oct 26 18:58 IMG_160729_073056_0348_RGB.JPG.jpg
-rw-r--r-- 1 root root 22691158 Oct 26 18:58 IMG_160729_073044_0343_RGB.JPG.jpg
-rw-r--r-- 1 root root 24510348 Oct 26 18:58 IMG_160729_073051_0346_RGB.JPG.jpg
-rw-r--r-- 1 root root 22615527 Oct 26 18:58 IMG_160729_073116_0353_RGB.JPG.jpg
-rw-r--r-- 1 root root 22111463 Oct 26 18:58 IMG_160729_073118_0354_RGB.JPG.jpg
-rw-r--r-- 1 root root 21955649 Oct 26 18:58 IMG_160729_073037_0340_RGB.JPG.jpg
-rw-r--r-- 1 root root 23847878 Oct 26 18:58 IMG_160729_073106_0349_RGB.JPG.jpg
drwxrwxrwt 1 root root     4096 Oct 26 18:58 .
-rw-r--r-- 1 root root 21550423 Oct 26 18:58 IMG_160729_073121_0355_RGB.JPG.jpg
-rw-r--r-- 1 root root 22491009 Oct 26 18:58 IMG_160729_073047_0344_RGB.JPG.jpg
-rw-r--r-- 1 root root 23223406 Oct 26 18:58 IMG_160729_073049_0345_RGB.JPG.jpg
-rw-r--r-- 1 root root 22232949 Oct 26 18:58 IMG_160729_073039_0341_RGB.JPG.jpg

dakotabenjamin, 2017-11-07:
I opened an issue since I just ran into the problem as well: https://github.com/nmoehrle/mvs-texturing/issues/89

dakotabenjamin, 2017-11-28:
OK So I have some insights about the size: These are the datasets photos being saved as pngs, so they don't have the jpg compression. EVEN THOUGH they say .jpg they're actually pngs.

dakotabenjamin, 2018-03-28:
This has been fixed in mvs-texturing and is reflecting in `master`

closed 2018-03-28 by dakotabenjamin