OpenDroneMap/ODM #703: Error in OdmMeshing: std::bad_alloc

tobi-kellner, 2017-10-29:
Dear everyone,
thank you for your work on this amazing project, apologies for taking your time with a newbie question.
I am running WebODM on a Wind 10 system using Docker and Oracle VM VirtualBox. My hardware is an i7-6700HQ (4 cores, 8 threads) with 16BG RAM, the VM is assigned 15 GB and 4 processors.
It can complete the sample projects without problem, but using my own bigger project (551 pics, 2.65GB in total) it failed 7h into the process.
The full output is here: 
[task_output.txt](https://github.com/OpenDroneMap/OpenDroneMap/files/1425460/task_output.txt)

I guess

> Error in OdmMeshing:
> std::bad_alloc

indicates that it ran out of RAM, though I am not sure if the preceding
> [WARNING] Initial residual too low: 7.92396e-08 < 0.000001

wants to tell me something.
I know the dataset itself works with ODM because I ran it successfully on an Amazon AWS EC2 instance with 32 GB of ram and 8 processors.

So, are there any paramters I can twiddle with to try and sort this out, ideally without sacrificing the quality of the final output?

When running ODM on AWS I was able to run it step-by-step so I could play with a problematic step without running from the very beginning, but it seems I cannot do that with WebODM...

Thank you very much for any help, greetings from Freiburg, Germany,

Tobi

dakotabenjamin, 2017-10-30:
What settings are you using for meshing?

dakotabenjamin, 2017-10-30:
Try lowering the --octree-depth to 10

tobi-kellner, 2017-10-30:
@dakotabenjamin thank you!
I will try the --octree-depth parameter (as soon as I have sorted out some file-space related issues in WebODM...).
Would increasing --mesh-solver-divide also help? If so, what is a reasonable increase - I probably ran at the default value of 9.

Thank you so much for your help!

closed 2018-03-28 by dakotabenjamin