OpenDroneMap/ODM #704: Unable to install opendrone map under ubuntu 16.0.5

peteraftsamy, 2017-10-30:
Hi,
am trying to install the opendronemap under ubuntu 16.0.5  however each time am ending up with 
----------------
ImportError: No module named csfm
Traceback (most recent call last):
  File "/home/peter/OpenDroneMap-0.3.1/run.py", line 46, in <module>
    plasm.execute(niter=1)
  File "/home/peter/OpenDroneMap-0.3.1/scripts/opensfm.py", line 85, in process
    (context.pyopencv_path, context.opensfm_path, tree.opensfm))
  File "/home/peter/OpenDroneMap-0.3.1/opendm/system.py", line 28, in run
    raise Exception("Child returned {}".format(retcode))
Exception: Child returned 1
peter@ajay-pc:~/OpenDroneMap-0.3.1$ 

Could u pl. advise

dakotabenjamin, 2017-10-30:
Hi,

Occasionally when installing some modules will fail to install due to some race conditions. Try running the installer again to see if opensfm installs properly. Let me know if that doesn't work.

peteraftsamy, 2017-10-31:
tried more than 3 times....but unable to proceed....

dakotabenjamin, 2017-10-31:
OK, can you clarify what operating system you are using? Ubuntu 16.0.5 is not a version that exists. 

Can you also find any errors in the configure output? That will help too.

dakotabenjamin, 2018-03-28:
Due to inactivity I will close this issue, please reopen if the problem is persisting.

closed 2018-03-28 by dakotabenjamin