OpenDroneMap/ODM #706: Crash in system.py

lorenzo93, 2017-11-07:
I ran the Docker container provided by you and at some point it crashes as showed by the screenshot.
Is there any fix?

[image]

dakotabenjamin, 2017-11-07:
I think you have to increase the memory of the docker instance

lorenzo93, 2017-11-07:
I gave docker 20GB of RAM, is it not enough?

dakotabenjamin, 2017-11-07:
No that should be enough. Can you give me more information? I see you're running docker in Windows. What was your run command?

lorenzo93, 2017-11-07:
The one provided in your guide, modified with my paths:

docker run -it --rm -v D:\images:/code/images -v D:\odm_orthophoto:/code/odm_orthophoto -v D:\odm_texturing:/code/odm_texturing opendronemap/opendronemap

The model it is very big, 370 pictures 3000x4000 each

pierotofy, 2017-11-10:
@dakotabenjamin I wonder if the addition of TBB https://github.com/nmoehrle/mvs-texturing/commit/313f20379bcca7db90bbcae636224ffb94252161 has caused memory usage to go up. I've seen quite a bit of reports like this in the last few weeks.

pierotofy, 2017-11-10:
I think we're seeing this happening more often because the images are pulled in their original definition (they are not resized, since we have removed the resize module).

The input images come from opensfm's `undistort` directory, and the files used to be resized.

pierotofy, 2017-11-10:
There's a linear correlation between image size and memory usage in the texrecon program.

A 170 image dataset at 4000px (largest side) used about 9GB of memory while running. The same dataset scaled to 2000px used about 5GB. We could consider re-adding the resize module to reduce memory usage.

closed 2020-04-14 by smathermather