OpenDroneMap/ODM #707: Coordinate string out of reference?

MilesSpathelf, 2017-11-09:
Hello,

I'm analyzing some drone imagery in Alaska and I came across this error:

[INFO]    Running ODM Texturing Cell - Finished
[INFO]    Running ODM Georeferencing Cell
[DEBUG]   None
[WARNING] No coordinates file. Generating coordinates file: /var/www/data/eab317a2-2d67-4521-8fd0-1da59970b125/odm_georeferencing/coords.txt

[INFO]    Running georeferencing with generated coords file.
[DEBUG]   running /code/build/bin/odm_georef -bundleFile /var/www/data/eab317a2-2d67-4521-8fd0-1da59970b125/opensfm/bundle_r000.out -inputCoordFile /var/www/data/eab317a2-2d67-4521-8fd0-1da59970b125/odm_georeferencing/coords.txt -inputFile /var/www/data/eab317a2-2d67-4521-8fd0-1da59970b125/odm_texturing/odm_textured_model.obj -outputFile /var/www/data/eab317a2-2d67-4521-8fd0-1da59970b125/odm_texturing/odm_textured_model_geo.obj -inputPointCloudFile /var/www/data/eab317a2-2d67-4521-8fd0-1da59970b125/pmvs/recon0/models/option-0000.ply -outputPointCloudFile /var/www/data/eab317a2-2d67-4521-8fd0-1da59970b125/odm_georeferencing/odm_georeferenced_model.ply  -logFile /var/www/data/eab317a2-2d67-4521-8fd0-1da59970b125/odm_georeferencing/odm_georeferencing_log.txt -outputTransformFile /var/www/data/eab317a2-2d67-4521-8fd0-1da59970b125/odm_georeferencing/odm_georeferencing_transform.txt -georefFileOutputPath /var/www/data/eab317a2-2d67-4521-8fd0-1da59970b125/odm_georeferencing/odm_georeferencing_model_geo.txt

  File "/code/run.py", line 47, in <module>
    plasm.execute(niter=1)
  File "/code/scripts/odm_georeferencing.py", line 162, in process
    geo_ref.parse_coordinate_system(tree.odm_georeferencing_coords)
  File "/code/opendm/types.py", line 317, in parse_coordinate_system
    self.utm_pole = ref[2][len(ref) - 1]
IndexError: string index out of range

It seems as if my coordinates are out of range?  Is Alaska not supported for ODM?
[task_output.txt](https://github.com/OpenDroneMap/OpenDroneMap/files/1459107/task_output.txt)

dakotabenjamin, 2017-11-09:
Hi, this is a duplicate of #698

MilesSpathelf, 2017-11-09:
Oh, ok thanks.  I didn't see that one.  So this is an ongoing error?

dakotabenjamin, 2017-11-09:
Yeah, check out here: https://github.com/OpenDroneMap/WebODM/issues/303 the fix has been added to the master branch

MilesSpathelf, 2017-11-09:
Hmm... I just updated via the WebODM interface before I started the run.  Is there another update I have to intstall?

dakotabenjamin, 2017-11-09:
you probably have to update your node-odm as well.

MilesSpathelf, 2017-11-09:
I apologize for my ignorance...  here is my node status as found through the WebODM interface

Hostname | node-odm-1
-- | --
Port | 3000
API Version | 1.0.3
Queue Count | 0
Last Refreshed | 0 minutes ago (Nov. 9, 2017, 8:56 p.m.)

pierotofy, 2017-11-09:
Hi @MilesSpathelf, perhaps this is a different bug; any chance you could share the dataset (or a subset of it) either privately or publicly via Dropbox/Google Drive?

pierotofy, 2017-11-10:
Can confirm, this is a bug.

Code will fail with `WGS84 UTM 6N`:

```
            line = 'WGS84 UTM 6N'
            log.ODM_DEBUG('Line: %s' % line)
            ref = line.split(' ')
            # match_wgs_utm = re.search('WGS84 UTM (\d{1,2})(N|S)', line, re.I)
            if ref[0] == 'WGS84' and ref[1] == 'UTM':  # match_wgs_utm:
                self.datum = ref[0]
                self.utm_pole = ref[2][len(ref) - 1]
```

Incriminating line is `ref[2][len(ref) - 1]`. Will open a PR for this by tomorrow.

MilesSpathelf, 2017-11-10:
Ah, many thanks for the confirmation!  Cheers,

Miles

closed 2017-11-20 by dakotabenjamin

MilesSpathelf, 2017-11-22:
Just wanted to say the update fixed my problem.  Many thanks!

pierotofy, 2017-11-22:
Ace 👍 you're welcome!

PR #708 (2017-11-20): Fixed utm_pole extraction
Fixes #707.

The reason this came under our radar just recently is that it affects a relatively small area of the world (all single digits UTM zones).

[image]

Just by coincidence `len(ref)` is usually `3` (just like `len('12S')`), but fails when the string is only two chars. 😄