OpenDroneMap/ODM #709: orthophoto distortion

bingzhang, 2017-11-13:
I use the default 20 pixel per meter and sometimes I found there is distortion in orthophoto. May I know there is option to remove distortion, please?

The line should be straight and continuous in the below image.
[image]

KommandorKeen, 2017-11-14:
@dakotabenjamin has code for undistorting images prior to use in ODM and ODM allows for undistortion to be turned off.  Does rely on you having a good camera calibration first though.

bingzhang, 2017-11-16:
@dakotabenjamin, May I have the code to try the undistorting images?

dakotabenjamin, 2017-11-28:
https://github.com/dakotabenjamin/cameracalibration

closed 2018-03-28 by dakotabenjamin