OpenDroneMap/ODM #710: Make GCP file specification case-insensitive

pierotofy, 2017-11-13:
If a user creates a GCP file, but fails to use the proper case when associating coordinates to filenames (which is common among Windows users where things are generally case insensitive), the odm_georef will happily discard all points that don't match a filename.

Not a fault, but we could make it more tolerant.

dmyers891258, 2017-11-13:
The wiki for including a GCP file has the file extensions in lowercase and they should be uppercase to work,

https://github.com/OpenDroneMap/OpenDroneMap/wiki/Running-OpenDroneMap

pierotofy, 2017-11-13:
Well, it depends on the dataset. The image filenames just have to match precisely.

I added a note for that in the Wiki.

yjmenezes, 2018-02-05:
My view point:

I think GCP name could be uppercase, as Windows can not deal with case properly.
PhotoName, images/* can be easily renamed to uppercase under UNIX-like systems;

A suggestion:
# for comment to turn point ON/OFF and an extra column with point identification.
It helps for large data sets.
 
I think it would be nice have a text file to replace ExifTags, for scanned or untagged images for example.

EPSG:32722
658528.361 7167002.098 890.636 2461 234 IMG_5349.JPG HV01
#658528.361 7167002.098 890.636 1824 1088 IMG_5350.JPG HV01
658528.361 7167002.098 890.636 2183 1898 IMG_5351.JPG HV01
658546.365 7166987.076 892.227 2436 336 IMG_5350.JPG HV05

pierotofy, 2021-09-27:
This hasn't been too much of an issue since most GCP files are now created using tools (e.g. POSM GCPi), so I will close this :pray: 

If there's still interest, we can re-open it.

closed 2021-09-27 by pierotofy