OpenDroneMap/ODM #711: What happens to the difference in the elevation of the Exif data? If I did several flights in the same project.

GeoCONeXion, 2017-11-14:
In one web page I read this:
"As of the December 2015 firmware releases DJI changed their JPG metadata format to use the GPS derived altitude instead of the relative barometric altimeter reading they had used previously. This is good news for general purpose geotagging since the above sea level number is now reflected. This is very bad news for users that are trying to use photogrammetry to accurately reconstruct a scene because the GPS derived altitude is notoriously noisy and unreliable."
Open Drone Map have this in count, went ODM used the exif data?. 
I did several flights in the same project and postproces this fotos with the ODM it take the elevation of the Exif data. I takeoff in the same place with the same elevation for each flight. Some times this elevation can have more than 15 meter of change betwen one photo to other. Is in the case of I did several flights in the same project.
What do you say about???

pierotofy, 2017-11-14:
Use GCPs if accuracy is important.

GeoCONeXion, 2017-11-14:
@pierotofy Yes, I know that. Put it is not the point.

pierotofy, 2017-11-14:
ODM at the moment has no way to know if a project has more than one flight. Would be an interesting concept to explore, however.

You shouldn't process all flights separately. If there are too many images to process all at once, keep an eye on the `split-merge` branch of the project for updates on processing very large datasets.

dakotabenjamin, 2017-11-28:
The scene warping is likely a camera issue. If you're using DJI, those cameras have large amounts of distortion. Try running camera calibration using this script: https://github.com/dakotabenjamin/cameracalibration

dakotabenjamin, 2018-03-28:
This issue was moved to http://community.opendronemap.org/t/what-happens-to-the-difference-in-the-elevation-of-the-exif-data-if-i-did-several-flights-in-the-same-project/365

closed 2018-03-28 by dakotabenjamin