OpenDroneMap/ODM #712: Running ODM on Ubuntu 16.04 not working

rionlerm, 2017-11-15:
Hi guys am I doing something wrong here?

I passed <bash configure.sh reinstall> and followed the _Readme_ instructions to run ODM.

ODM is installed on removable media and ENVs are set to an empty directory (path to project) on the Desktop,

When I pass <./run.sh odm_data_aukerman> (the data dir. with images folder is inside the Desktop projects path as a sub-directory but the Terminal tells me: <bash: ./run.sh: Permission denied>

What could be the problem?

Wes-Skeeter, 2017-11-24:
Why did you run reinstall instead of install, did you have a previous version working at some point?  Have you tried running `sudo ./run.sh` ?

dakotabenjamin, 2017-11-28:
Do you normally run on removable media? You may have to make sure that the user logged into the computer you are running from has permission to run `run.sh`

You can check permissions with `ls -l`

rionlerm, 2017-12-06:
@Wes-Skeeter I had a previous version working. Passing `sudo run.sh` or `bash run.sh` the missing ecto error pops up.

@dakotabenjamin I normally don't run on removable media, but even after ...

dakotabenjamin, 2017-12-06:
It's just a guess but trying to run from the removable media might be the problem.

rionlerm, 2017-12-07:
Cool, I shall do a reinstall from the Desktop.

Regards,
*Rion Lerm (jnr.)*
(+27) 76 913 8381 (Cell)
(+27) 13 735 3536 (Office)
Ndlovu Node
Phalaborwa Gate, Kruger National Park
South African Environmental Observation Network (SAEON)

Click here to learn about SAEON <http://www.saeon.ac.za/saeon-background>

rionlerm, 2017-12-07:
@dakotabenjamin something that might point you in the right direction towards a solution. When running `bash configure.sh reinstall` from Desktop I noticed this:

[image]

rionlerm, 2017-12-07:
Is it maybe related to [Issue #217](https://github.com/OpenDroneMap/OpenDroneMap/issues/217)?

alke, 2017-12-16:
I changed in file configure.sh
from all rows has "sudo pip install"
to "sudo -H pip install"

rionlerm, 2018-01-04:
Hi @alke,

I made the change. Let's see what happens when I pass *bash configure.sh
reinstall*

...

Regards,
*Rion Lerm (jnr.)*
(+27) 76 913 8381 (Cell)
(+27) 13 735 3536 (Office)
Ndlovu Node
Phalaborwa Gate, Kruger National Park
South African Environmental Observation Network (SAEON)

Click here to learn about SAEON <http://www.saeon.ac.za/saeon-background>

rionlerm, 2018-01-04:
What's going on? Still not working. @dakotabenjamin ?

[image]

dakotabenjamin, 2018-01-04:
Something is throwing an error and stopping the build. Can you
```
cd Superbuild/build
cmake ..
make -j1
```
and see if you can catch the error being thrown?

rionlerm, 2018-01-05:
Hi @dakotabenjamin ,

see result below as per your instruction:

[image]

dakotabenjamin, 2018-01-08:
You mentioned before that you have a bit of a slow internet connection, right? It might be timing out...

rionlerm, 2018-01-09:
Hi @dakotabenjamin ,

our internet now runs faster than before and I performed the last install with no staff at our offices that could decrease bandwidth. What I did change though is I installed Ubuntu 16. Since then as I recall it, doesn't want to install.

I am following the short install description for Ubuntu 16 on the main code page of the master branch.

yjmenezes, 2018-01-23:
@rionlerm 

I have compiled ODM 0.3.1 commit 77251dd under Ubuntu 16.04 x64 and configure.sh install runs fine for me.
I just change "sudo -H pip install" as @alke point out.
 
./run.sh --project-path   tests  test_data --use-exif 
it works as expected.

On the other hand, i have problems with my dataset.
I am trying to figure out where is the problem .

PhaseOne iXa180, 55mm two strips 10 photos + LiDAR.

I upload my dataset to git,  https://github.com/yjmenezes/church_dataset

I tryed:
./run.sh --project-path /hmnt/ church_data  --force-focal 55 --force-ccd 53.706 --resize-to 0 --use-25dmesh --use-exif

Maybe someone wanna test my church_dataset  and help me to fix my mistakes.

i apreciate any help.

btw, ODM crash with:
[DEBUG]   running PYTHONPATH=/hmnt/OpenDroneMap/SuperBuild/install/lib/python2.7/dist-packages /hmnt/OpenDroneMap/SuperBuild/src/opensfm/bin/opensfm create_tracks /hmnt/church_data/opensfm
2018-01-23 14:23:52,590 reading features
2018-01-23 14:23:53,974 Merging features onto tracks
2018-01-23 14:23:53,974 Good tracks: 0
[DEBUG]   running PYTHONPATH=/hmnt/OpenDroneMap/SuperBuild/install/lib/python2.7/dist-packages /hmnt/OpenDroneMap/SuperBuild/src/opensfm/bin/opensfm reconstruct /hmnt/church_data/opensfm
2018-01-23 14:23:56,233 Starting incremental reconstruction
2018-01-23 14:23:56,238 0 partial reconstructions in total.
[DEBUG]   running PYTHONPATH=/hmnt/OpenDroneMap/SuperBuild/install/lib/python2.7/dist-packages /hmnt/OpenDroneMap/SuperBuild/src/opensfm/bin/opensfm mesh /hmnt/church_data/opensfm
Traceback (most recent call last):
  File "/hmnt/OpenDroneMap/SuperBuild/src/opensfm/bin/opensfm", line 34, in <module>
    command.run(args)
  File "/hmnt/OpenDroneMap/SuperBuild/src/opensfm/opensfm/commands/mesh.py", line 22, in run
    reconstructions = data.load_reconstruction()
  File "/hmnt/OpenDroneMap/SuperBuild/src/opensfm/opensfm/dataset.py", line 368, in load_reconstruction
    with open(self.__reconstruction_file(filename)) as fin:
IOError: [Errno 2] No such file or directory: '/hmnt/church_data/opensfm/reconstruction.json'
Traceback (most recent call last):
  File "/hmnt/OpenDroneMap/run.py", line 47, in <module>
    plasm.execute(niter=1)
  File "/hmnt/OpenDroneMap/scripts/opensfm.py", line 131, in process
    (context.pyopencv_path, context.opensfm_path, tree.opensfm))
  File "/hmnt/OpenDroneMap/opendm/system.py", line 34, in run
    raise Exception("Child returned {}".format(retcode))
Exception: Child returned 1

rionlerm, 2018-01-24:
Thanks @yjmenezes ,

the problem still persists. I don't know what;s going on and support from other ODM experts are few and far between RE this issue.

I added "-H" to any pip install areas of the configure.sh file by amending it with the gedit command

[image]

rionlerm, 2018-01-24:
Maybe it's a Time-out that never happened before even with slower bandwidth. Nonetheless, here's the second install's results that look promising:

[image]

dakotabenjamin, 2018-01-24:
Apologies, working on deadlines on this end. Would this issue be relevant? https://github.com/OpenDroneMap/OpenDroneMap/issues/695

rionlerm, 2018-01-25:
Thanks @dakotabenjamin , it seems that issue is relevant.

You will hear from me if things go pear-shaped again.

Regards,

rionlerm, 2018-01-25:
So I deleted the ODM folder from my PC and passed `bash configure.sh reinstall` under the downloaded 0.3.1 version of ODM

It still ends with PCL causing probs.:

[image]

rionlerm, 2018-01-25:
Followed issue #695 and and changed to "sudo -H pip install"...and the result is:

[image]

yjmenezes, 2018-01-29:
@rionlerm 

I repeat the ./configure.sh install 6 over a fresh Ubuntu 16.06x64 desktop  30GBram  QEMU guest.
The ODM 0.3.1, commit #be4ddf5  /configure.sh install  works for me.

I think the problem can really be some time-out during downloads.

Unfortunately, my git church dataset is crashing as before.

We go ahead, ODM is a great work, thank you  ODM developers team.

dakotabenjamin, 2018-03-28:
This issue was moved to http://community.opendronemap.org/t/building-running-odm-on-ubuntu-16-04-not-working/382

closed 2018-03-28 by dakotabenjamin