OpenDroneMap/ODM #714: Error with Docker

florianbeyer, 2017-11-17:
Hi,

I'm using docker to perform ODM.

`sudo docker run -it --rm -v ~/Documents/ODM_projects/odm_tribsees/images:/code/images -v ~/Documents/ODM_projects/odm_tribsees/odm_orthophoto:/code/odm_orthophoto -v ~/Documents/ODM_projects/odm_tribsees/odm_texturing:/code/odm_texturing opendronemap/opendronemap
`

It takes a while and process the data.
During processing I got the following error:
```
	Writing data cost file... done.
	2746 faces have not been seen by a view.
	Comp	Iter	Energy		Runtime
	0000	0	278335	0.008
	0000	1	206246	0.665
	0000	2	186136	1.307
	0000	3	177782	1.949
	0000	4	172069	2.606
	0000	5	167856	3.263
	0000	6	165498	3.922
	0000	7	163897	4.57
	0000	8	162838	5.215
	0000	9	162258	5.913
	0000	10	161384	6.566
	0000	11	160591	7.219
	0000	12	159872	7.868
	0000	13	159027	8.524
	0000	14	158662	9.175
	0000	15	158049	9.829
	0000	16	157522	10.481
	0000	17	157276	11.126
	0000	18	156889	11.772
	0000	19	156584	12.418
	0000	20	156372	13.076
	0000	21	156045	13.733
	0000	22	155968	14.385
	0000	23	155587	15.034
	0000	24	155439	15.68
	0000	25	155385	16.334
	0000	26	155107	16.978
	0000	27	154959	17.624
	0000	28	154687	18.266
	0000	29	154484	18.91
	0000	30	154478	19.565
	0000	31	154373	20.226
	0000	32	154288	20.936
	0000	33	154166	21.588
	0000	34	154119	22.244
	0000	35	153995	22.893
	0000	36	153987	23.535
	0000	37	153906	24.192
	0000	38	153851	24.845
	0000	39	153819	25.496
	0000	40	153642	26.14
	0000	41	153552	26.785
	0000	42	153572
	0000	Increase of energy - stopping optimization
	Took: 262.069s
Generating texture patches:
	Running... done. (Took 378.718s)
	368 texture patches.
Running global seam leveling:
	Create matrices for optimization...  done.
	Lhs dimensionality: 105836 x 105836
	Calculating adjustments:
		Color channel 0: CG took 88 iterations. Residual is 9.02522e-05
		Color channel 1: CG took 88 iterations. Residual is 9.62096e-05
		Color channel 2: CG took 83 iterations. Residual is 9.27548e-05
		Took 1.645 seconds
        Adjusting texture patches 0%...Killed                                   
Traceback (most recent call last):
  File "/code/run.py", line 47, in <module>
    plasm.execute(niter=1)
  File "/code/scripts/mvstex.py", line 124, in process
    '{keepUnseenFaces}'.format(**kwargs))
  File "/code/opendm/system.py", line 34, in run
    raise Exception("Child returned {}".format(retcode))
Exception: Child returned 137
```

Can someone help me?
What is the problem?

dakotabenjamin, 2017-11-28:
Likely you are running out of memory.

pierotofy, 2017-11-29:
Yup, `137` means out of memory.

dakotabenjamin, 2018-03-28:
This issue was moved to http://community.opendronemap.org/t/error-with-docker/389

closed 2018-03-28 by dakotabenjamin