OpenDroneMap/ODM #715: DEM generation option is ambiguous

Wes-Skeeter, 2017-11-17:
I'm trying to create a DEM, but when I run the following command:

`ubuntu@ip-172-31-7-75:~/OpenDroneMap$ ./run.sh -i /home/ubuntu/Temp_Data 821_Down_and_Oblique_HighRes_wDEM --opensfm-processes 8 --min-num-features 10000 --dem --use-25dmesh`

I get this traceback
```
usage: run.py [options] <project name>
run.py: error: ambiguous option: --dem could match --dem-gapfill-steps, --dem-resolution, --dem-maxsd, --dem-approximate, --dem-terrain-type, --dem-maxangle, --dem-initial-distance, --dem-decimation
```

dakotabenjamin, 2017-11-20:
You will want to use either `--dtm` or `--dsm` instead.

<http://www.clemetparks.com/>

dakotabenjamin, 2017-11-28:
Closing this, please reopen if you still have issues.

closed 2017-11-28 by dakotabenjamin