OpenDroneMap/ODM #716: Error on running odm -- 'No Module named 'appsettings'

leeprevost, 2017-11-18:
I created an odm image using option 2 replication from github.  

On my first run, I entered the command line as instructed and got the following traceback.

> Traceback (most recent call last):
>   File "run.py", line 4, in <module>
>     from opendm import config
>   File "C:\Users\lprevost\odm\OpenDroneMap\opendm\config.py", line 6, in <module>
>     from appsettings import SettingsParser
> ModuleNotFoundError: No module named 'appsettings'

dakotabenjamin, 2017-11-28:
Can you try reinstalling? `./configure.sh reinstall 1`

dakotabenjamin, 2018-03-28:
This issue was moved to http://community.opendronemap.org/t/error-on-running-odm-no-module-named-appsettings/396

closed 2018-03-28 by dakotabenjamin

stevebroskey, 2018-10-11:
I have this same issue.
Ubuntu, 16.04.5 LTS; following the instructions [here](https://github.com/OpenDroneMap/OpenDroneMap#native-install-ubuntu-1604).
I'm attempting the reinstall right now.

Three points of inquiry:
1. I had to install with `sudo` (`sudo bash configure.sh install`), or the installation failed with permissions errors. Is this going to cause errors later? (I may have used the poor practice of [`sudo pip...`](https://askubuntu.com/a/802594/182241)
2. In the above-linked-to quickstart, the instructions include `bash` for initiating the install script. Is this necessary, since elsewhere in the docs it simply says `./configure.sh` (vs `bash ./configure.sh`)
3. Does `reinstall` require the `1`, or is it sufficient to execute `sudo ./configure.sh reinstall`?

smathermather, 2018-10-11:
Hi @stevebroskey -- this is a pretty old issue, and likely a better place to get help on this is https://community.opendronemap.org. Thanks!

stevebroskey, 2018-10-11:
> Hi @stevebroskey -- this is a pretty old issue, and likely a better place to get help on this is https://community.opendronemap.org. Thanks!

Thanks; JFYI the https in your link broke the link.

smathermather, 2018-10-11:
Ugh. We should add a cert... . Thanks for the catch.