OpenDroneMap/ODM #719: Lots of distortion in high buildings

ivanatora, 2017-11-27:
Hi, guys.

First attempt at ortophoto with OpenDroneMap.

I've just compared results with DroneDeploy and it seems ODM produces way worse imagery for buildings in ortophoto. I'm using the docker version of ODM. These buildings are ~30m high and drone altitude was ~100m.

Here is comparison:

OpenDroneMap result:
https://i.imgur.com/ZIuhpH5.jpg

DroneDeploy result:
https://i.imgur.com/hdV2zMb.jpg

Can ODM do better with paralax distortions?

dakotabenjamin, 2017-11-27:
Try rerunning using `--rerun-from odm_meshing --use-25dmesh`

There will still be some distortion around the edges. It's a known problem with our meshing algorithm and we are working on it.

ivanatora, 2017-11-27:
Thanks for your reply! How do I add custom parameters to the docker command?

ivanatora, 2017-11-27:
Thanks for your reply! How do I add custom parameters to the docker command? Just append it at the end like this?
> $ docker run -it --rm -v $(pwd)/images:/code/images -v $(pwd)/odm_orthophoto:/code/odm_orthophoto -v $(pwd)/odm_texturing:/code/odm_texturing opendronemap/opendronemap --rerun-from odm_meshing --use-25dmesh`

dakotabenjamin, 2017-11-27:
Ah I should have asked if you were running docker. The --rerun-from param is useless, unfortunately. I would recommend linking the opensfm folder: `-v $(pwd)/opensfm:/code/opensfm` to save you time on future reruns. But yes, that is how you pass parameters.

ivanatora, 2017-11-28:
Let it run for the night. I'm having mixed feelings about the result. The high buildings are a bit better. But the crane and the lower structures are curly and there are white spots on the ground.

Here is the result:
https://i.imgur.com/9m5EkWO.jpg

ivanatora, 2017-11-28:
I've read a bit about drone mapping and people recommend flying altitude to be at least 4-5 times higher than the highest object in the mapped area. If this is the case 30 m buildings are too high for my flying altitude of 100 m and it is my fault and not ODM's. I will try to re-fly the project in a few days.

bingzhang, 2017-11-28:
mark it.

closed 2017-11-28 by ivanatora