OpenDroneMap/ODM #721: 3D Projection is generating images with a line

juvinski, 2017-11-28:
Hi,

Processing images are resulting in random lines on the final image like this:
[image]
Or
[image]
Already change the texturing-data-term but still having problem. 
I see the problem on 3d model too.

pierotofy, 2017-11-29:
I've seen this problem before on other datasets, unfortunately I'm not familiar enough with the mve_texturing module to say for sure what is the underlying cause. It might be worth to open an issue on https://github.com/nmoehrle/mvs-texturing and ask for help there (at least to get a direction on the possible cause).

juvinski, 2017-11-30:
Hi @pierotofy 
I will do that right now

juvinski, 2019-02-18:
Hi,

just an update. After some tests and research, I could map the main reason of this problem is something on my images. If I run on an images just after the camera shot the pictures, I had this behavior, if I run something to correct the image distortion before the processing, the result is with no lines.

smathermather, 2019-02-22:
@juvinski -- that makes sense. Undistorting would improve the ability of mvs-texturing to map the textures substantially.

I'll mark this as a duplicate of https://github.com/OpenDroneMap/ODM/issues/538 then and close. Thanks for revisiting this!

closed 2019-02-22 by smathermather