OpenDroneMap/ODM #722: Feature Request: remove spaces from files before processing them

fredlllll, 2017-11-28:
odm or some of its libraries have problems if the images filename contains spaces. so replacing those spaces before processing would be a good idea

pierotofy, 2017-11-29:
Does processing fail when there's a space? What is the output from the log?

fredlllll, 2017-11-29:
yeah processing fails at the texturing step. the nvm scene format seems to be space seperated, and if you put a filename with a space in there, the program cant handle that and crashes. of course with a less than helpful error. it basically tells you it tried ot open a folder as an image

pierotofy, 2017-11-29:
Could you copy/paste the full output here? Perhaps if it affects only the texturing step we could fix the problem there.

fredlllll, 2017-11-29:
```
[INFO] Found 835 usable images [INFO] Running ODM Load Dataset Cell - Finished [INFO] Running ODM OpenSfM Cell [WARNING] Found a valid OpenSfM reconstruction file in: /srv/workspace/proj_9053564971/opensfm/reconstruction.json [WARNING] Found a valid Bundler file in: /srv/workspace/proj_9053564971/opensfm/reconstruction.json [INFO] Running ODM OpenSfM Cell - Finished [INFO] Running ODM Meshing Cell [WARNING] Found a valid ODM Mesh file in: /srv/workspace/proj_9053564971/odm_meshing/odm_mesh.ply [INFO] Running ODM Meshing Cell - Finished [INFO] Running MVS Texturing Cell [DEBUG] Writing MVS Textured file in: /srv/workspace/proj_9053564971/odm_texturing/odm_textured_model.obj [DEBUG] running /srv/OpenDroneMap/SuperBuild/install/bin/texrecon /srv/workspace/proj_9053564971/opensfm/reconstruction.nvm /srv/workspace/proj_9053564971/odm_meshing/odm_mesh.ply /srv/workspace/proj_9053564971/odm_texturing/odm_textured_model -d gmi -o gauss_clamping -t none /srv/OpenDroneMap/SuperBuild/install/bin/texrecon (built on Nov 28 2017, 11:01:57) Load and prepare mesh: PLY Loader: comment PCL generated Reading PLY: 100002 verts... 199767 faces... done. Generating texture views: NVM: Loading file... NVM: Number of views: 501 NVM: Number of features: 0 Loading 0%...terminate called after throwing an instance of 'util::Exception' terminate called recursively terminate called recursively terminate called recursively what(): /srv/workspace/proj_9053564971/opensfm/: Cannot determine image format Aborted (core dumped) Traceback (most recent call last): File "/srv/OpenDroneMap/run.py", line 47, in <module> plasm.execute(niter=1) File "/srv/OpenDroneMap/scripts/mvstex.py", line 124, in process '{keepUnseenFaces}'.format(**kwargs)) File "/srv/OpenDroneMap/opendm/system.py", line 34, in run raise Exception("Child returned {}".format(retcode)) Exception: Child returned 134
```

dakotabenjamin, 2017-11-29:
I'm working on the temporary files issue right now and it's in the same part of the code so I can look into this

closed 2020-04-14 by smathermather