OpenDroneMap/ODM #723: wb docker image not working for me

matthewberryman, 2017-11-29:
Have tried on both [odm_data_waterbury](https://github.com/OpenDroneMap/odm_data_waterbury) and [odm_data_bellus](https://github.com/OpenDroneMap/odm_data_bellus) data sets with command `docker run -d -v $(pwd)/images:/code/images -v $(pwd)/odm_orthophoto:/code/odm_orthophoto -v $(pwd)/odm_texturing:/code/odm_texturing opendronemap/opendronemap:wb` and both fail with the following error
```
2017-11-29 03:40:24,155 Starting incremental reconstruction
Traceback (most recent call last):
  File "/code/SuperBuild/src/opensfm/bin/opensfm", line 34, in <module>
    command.run(args)
  File "/code/SuperBuild/src/opensfm/opensfm/commands/reconstruct.py", line 20, in run
    reconstruction.incremental_reconstruction(data)
  File "/code/SuperBuild/src/opensfm/opensfm/reconstruction.py", line 1000, in incremental_reconstruction
    pairs = compute_image_pairs(common_tracks, data)
  File "/code/SuperBuild/src/opensfm/opensfm/reconstruction.py", line 353, in compute_image_pairs
    result = parallel_map(_compute_pair_reconstructability, args, processes)
  File "/code/SuperBuild/src/opensfm/opensfm/context.py", line 35, in parallel_map
    return list(e.map(func, args))
  File "/usr/local/lib/python2.7/dist-packages/loky/process_executor.py", line 769, in _chain_from_iterable_of_lists
    for element in iterable:
  File "/usr/local/lib/python2.7/dist-packages/loky/_base.py", line 584, in result_iterator
    yield future.result()
  File "/usr/local/lib/python2.7/dist-packages/loky/_base.py", line 431, in result
    return self.__get_result()
  File "/usr/local/lib/python2.7/dist-packages/loky/_base.py", line 382, in __get_result
    raise self._exception
loky.process_executor.BrokenProcessPool: A process in the process pool was terminated abruptly while the future was running or pending.
Traceback (most recent call last):
  File "/code/run.py", line 47, in <module>
    plasm.execute(niter=1)
  File "/code/scripts/opensfm.py", line 115, in process
    (context.pyopencv_path, context.opensfm_path, tree.opensfm))
  File "/code/opendm/system.py", line 34, in run
    raise Exception("Child returned {}".format(retcode))
Exception: Child returned 1
```

This is on a box with 32 AMD cores and 64 GB of RAM

dakotabenjamin, 2017-11-30:
@pierotofy have you seen this before? 

I found a [stack overflow thread](https://stackoverflow.com/questions/15900366/all-example-concurrent-futures-code-is-failing-with-brokenprocesspool#15918742) but I'm not sure how to understand it. is the issue that opensfm's reconstruction.py doesn't have an `if __name__ == "__main__"`?

pierotofy, 2017-11-30:
How was the `wb` image built? I'm not sure about the cause.

dakotabenjamin, 2017-11-30:
It pulls latest opensfm for flat scene initialization: https://github.com/OpenDroneMap/OpenDroneMap/tree/wb

dakotabenjamin, 2017-11-30:
If you run with --opensfm-processes 1, you get a different error: 
```
[INFO]    Running ODM OpenSfM Cell
[DEBUG]   Altitude data detected, enabling it for GPS alignment
[WARNING] Found a feature matching done progress file in: /code/opensfm/matching_done.txt
[WARNING] Found a valid OpenSfM tracks file in: /code/opensfm/tracks.csv
[DEBUG]   running PYTHONPATH=/code/SuperBuild/install/lib/python2.7/dist-packages /code/SuperBuild/src/opensfm/bin/opensfm reconstruct /code/opensfm
2017-11-30 16:22:26,792 Starting incremental reconstruction
Illegal instruction (core dumped)
Traceback (most recent call last):
  File "/code/run.py", line 47, in <module>
    plasm.execute(niter=1)
  File "/code/scripts/opensfm.py", line 115, in process
    (context.pyopencv_path, context.opensfm_path, tree.opensfm))
  File "/code/opendm/system.py", line 34, in run
    raise Exception("Child returned {}".format(retcode))
Exception: Child returned 132
```

pierotofy, 2017-11-30:
Illegal instruction means an executable contains an invalid instruction. This happens when you build a docker image on a certain architecture and then try to run the same docker image on a different machine with a different architecture (or CPU).

Not a problem with opensfm. If you want to create portable images, you need to turn off all optimizations (see `core2.Dockerfile` in the OpenDroneMap repository).

pierotofy, 2017-11-30:
https://github.com/OpenDroneMap/OpenDroneMap/blob/master/core2.Dockerfile#L48

dakotabenjamin, 2017-11-30:
AH ok.

dakotabenjamin, 2017-11-30:
@matthewberryman I will rebuild and push, then test on the server.

dakotabenjamin, 2017-11-30:
The `wb` image is now working.

closed 2017-11-30 by dakotabenjamin