OpenDroneMap/ODM #727: Trying to create a DSM

OrangOrangUtan, 2017-12-18:
Running OpenDroneMap using a prebuilt Docker image on Mac OS X 10.13. I am testing out this program using the "aukerman" example data (https://github.com/OpenDroneMap/odm_data_aukerman). Things run fine until:
```
Running global seam leveling:
	Create matrices for optimization...  done.
	Lhs dimensionality: 109804 x 109804
	Calculating adjustments:
		Color channel 0: CG took 108 iterations. Residual is 9.4721e-05
		Color channel 1: CG took 107 iterations. Residual is 9.92034e-05
		Color channel 2: CG took 111 iterations. Residual is 9.82466e-05
		Took 1.061 seconds
        Adjusting texture patches 16%...Killed 
Traceback (most recent call last):
  File "/code/run.py", line 47, in <module>
    plasm.execute(niter=1)
  File "/code/scripts/mvstex.py", line 124, in process
    '{keepUnseenFaces}'.format(**kwargs))
  File "/code/opendm/system.py", line 34, in run
    raise Exception("Child returned {}".format(retcode))
Exception: Child returned 137
```
I know this must be because of my computer's small amount of RAM (4 GB, all of which I've allocated to Docker). My friend has tried running the same process on his Windows machine (8 GB RAM) and runs into the same error at the same spot. 

Our ultimate goal is to generate a DSM to look at canopy height in a forested area; thus, we're not specifically interested in 3D models, orthomosaics, or anything else OpenDroneMap can produce other than the DSM. **Is there any way to bypass the texturing step, and/or any other steps necessary for producing specifically the 3d model/orthomosaic, in order to make the entire process require less memory?**

dakotabenjamin, 2017-12-18:
You can try resizing the images using `--resize-to` which will make mvs-texturing less memory intensive. We are working on building some turn-key parameters for smaller machines and also looking at ways to simplify the process for users who only want specific outputs.

OrangOrangUtan, 2017-12-18:
ok I will try resizing!

dakotabenjamin, 2018-03-28:
Did this work? If not please reopen.

closed 2018-03-28 by dakotabenjamin