OpenDroneMap/ODM #729: Background color of the Orthophoto is always (255, 255, 255, 0)

jtinguely, 2017-12-19:
Expose a parameter to specify the background color for an Orthophoto.

theoway, 2022-08-09:
@pierotofy We can either create a standalone script for this and add it to `contrib` folder. Or we can do this in `postprocessing` step, here:
https://github.com/OpenDroneMap/ODM/blob/5d54d342c9d562b259a3f790aeb7898e251da639/opendm/orthophoto.py#L88
Let me know your thoughts and I'll make a PR accordingly.

pierotofy, 2022-08-09:
I think the idea here was to add a different default color in the `odm_orthophoto` program, but I'm not even sure this is really needed anymore? (all orthophotos have a transparency band so one can add whatever background they want post ODM). I wonder if this can be closed.

closed 2022-08-10 by pierotofy