OpenDroneMap/ODM #730: Segmentation fault at georeferencing step

mjevanmarle, 2017-12-20:
I am nearly at the end of a webodm process when the scripts get stuck at georeferencing. The provided testphotos work properly, but using my own photos without GCP file results in an segmentation fault and using a GCP file results in a general failure without specified error. I would be grateful if anyone can help me. Both get stuck at the georeferencing.

Please find the error without GCP file below. Test files can be found here: https://www.dropbox.com/sh/vgof70820mbea4v/AABMVchOGVD6o_ZjaRGgCc5ta?dl=0 

[WARNING] No coordinates file. Generating coordinates file: /opt/odm/node-OpenDroneMap/data/5156a2cf-47df-463f-b990-c6e6bdb8f641/odm_georeferencing/coords.txt
[INFO]    Running georeferencing with generated coords file.
[DEBUG]   running /opt/odm/OpenDroneMap_v0_3_1/build/bin/odm_georef -bundleFile /opt/odm/node-OpenDroneMap/data/5156a2cf-47df-463f-b990-c6e6bdb8f641/opensfm/bundle_r000.out -inputCoordFile /opt/odm/node-OpenDroneMap/data/5156a2cf-47df-463f-b990-c6e6bdb8f641/odm_georeferencing/coords.txt -inputFile /opt/odm/node-OpenDroneMap/data/5156a2cf-47df-463f-b990-c6e6bdb8f641/odm_texturing/odm_textured_model.obj -outputFile /opt/odm/node-OpenDroneMap/data/5156a2cf-47df-463f-b990-c6e6bdb8f641/odm_texturing/odm_textured_model_geo.obj -inputPointCloudFile /opt/odm/node-OpenDroneMap/data/5156a2cf-47df-463f-b990-c6e6bdb8f641/opensfm/depthmaps/merged.ply -outputPointCloudFile /opt/odm/node-OpenDroneMap/data/5156a2cf-47df-463f-b990-c6e6bdb8f641/odm_georeferencing/odm_georeferenced_model.ply  -logFile /opt/odm/node-OpenDroneMap/data/5156a2cf-47df-463f-b990-c6e6bdb8f641/odm_georeferencing/odm_georeferencing_log.txt -georefFileOutputPath /opt/odm/node-OpenDroneMap/data/5156a2cf-47df-463f-b990-c6e6bdb8f641/odm_georeferencing/odm_georeferencing_model_geo.txt
Segmentation fault (core dumped)
Traceback (most recent call last):
  File "/opt/odm/OpenDroneMap/run.py", line 46, in <module>
    plasm.execute(niter=1)
  File "/opt/odm/OpenDroneMap_v0_3_1/scripts/odm_georeferencing.py", line 137, in process
    '-logFile {log} -georefFileOutputPath {geo_sys}'.format(**kwargs))
  File "/opt/odm/OpenDroneMap_v0_3_1/opendm/system.py", line 28, in run
    raise Exception("Child returned {}".format(retcode))
Exception: Child returned 139

pierotofy, 2017-12-22:
Not sure if it's the cause, but 20171114-105312_DJI_FC300X_DJI_0046.JPG shows as corrupted.

mjevanmarle, 2017-12-22:
Thanks @pierotofy but excluding the 0046.jpg does not make a difference. It still gets stuck at the georefencing part. Do you maybe have a clue of a module that needs to be reinstalled or which one could be involved in this issue?

pierotofy, 2017-12-22:
Looks like your images' exif tags are invalid. For example:

```bash
# exiftool 20171114-105311_DJI_FC300X_DJI_0045.JPG 
GPS Latitude                    : 0 deg 0' 0.00" N
GPS Longitude                   : 0 deg 0' 0.00" E
GPS Position                    : 0 deg 0' 0.00" N, 0 deg 0' 0.00" E
```

dakotabenjamin, 2018-03-28:
@mjevanmarle can you confirm piero's suggestion was the cause? Please reopen if not

closed 2018-03-28 by dakotabenjamin