OpenDroneMap/ODM #731: No such file or directory?

OrangOrangUtan, 2017-12-21:
I am running OpenDroneMap on Ubuntu 16.04 (on a machine with 32 GB RAM) after no luck using a Macbook Air. Running into the following issue:
```
Generating texture views: 
NVM: Loading file...
NVM: Number of views: 75
NVM: Number of features: 0
        Loading 0%...terminate called after throwing an instance of 'util::FileException'
  what():  No such file or directory
Aborted (core dumped)
Traceback (most recent call last):
  File "/code/run.py", line 47, in <module>
    plasm.execute(niter=1)
  File "/code/scripts/mvstex.py", line 124, in process
    '{keepUnseenFaces}'.format(**kwargs))
  File "/code/opendm/system.py", line 34, in run
    raise Exception("Child returned {}".format(retcode))
Exception: Child returned 134
```

I tried adding the setting `--use-opensfm-pointcloud` in accordance with <https://github.com/OpenDroneMap/OpenDroneMap/issues/486>. It didn't seem to work: 
```
sudo docker run -it --rm -v $(pwd)/images:/code/images -v $(pwd)/odm_orthophoto:/code/odm_orthophoto -v $(pwd)/odm_texturing:/code/odm_texturing opendronemap/opendronemap --use-opensfm-pointcloud
sudo: unable to resolve host ip-172-30-0-212
usage: run.py [options] <project name>
run.py: error: unrecognized arguments: --use-opensfm-pointcloud
```

pierotofy, 2018-01-03:
Hey @OrangOrangUtan, could you copy paste the entire output from the console? Perhaps something went wrong with an earlier stage of the process.

dakotabenjamin, 2018-03-28:
This issue was moved to http://community.opendronemap.org/t/no-such-file-or-directory/384

closed 2018-03-28 by dakotabenjamin