OpenDroneMap/ODM #734: OpenDroneMap failing on "Garmin VIRB" exif data:

bearyjd, 2017-12-30:
I am having problems with Garmin Virb EXIF data as well.  I believe that this was listed in issue 104.

https://github.com/OpenDroneMap/OpenDroneMap/issues/104

Whenever I run WebODM with VIRB photos, I get the following:

--------------------+----------------------------------------------------------
2017-12-30 19:27:23,801 Ceres Solver Report: Iterations: -2, Initial cost: -1.000000e+00, Final cost: -1.000000e+00, Termination: FAILURE
Traceback (most recent call last):
File "/code/SuperBuild/src/opensfm/bin/opensfm", line 34, in 
command.run(args)
File "/code/SuperBuild/src/opensfm/opensfm/commands/reconstruct.py", line 21, in run
report = reconstruction.incremental_reconstruction(data)
File "/code/SuperBuild/src/opensfm/opensfm/reconstruction.py", line 1069, in incremental_reconstruction
data, graph, reconstruction, remaining_images, gcp)
File "/code/SuperBuild/src/opensfm/opensfm/reconstruction.py", line 1031, in grow_reconstruction
align.align_reconstruction(reconstruction, gcp, data.config)
File "/code/SuperBuild/src/opensfm/opensfm/align.py", line 14, in align_reconstruction
res = align_reconstruction_similarity(reconstruction, gcp, config)
File "/code/SuperBuild/src/opensfm/opensfm/align.py", line 56, in align_reconstruction_similarity
return align_reconstruction_naive_similarity(reconstruction, gcp)
File "/code/SuperBuild/src/opensfm/opensfm/align.py", line 80, in align_reconstruction_naive_similarity
T = tf.superimposition_matrix(X.T, Xp.T, scale=True)
File "/code/SuperBuild/src/opensfm/opensfm/transformations.py", line 1041, in superimposition_matrix
scale=scale, usesvd=usesvd)
File "/code/SuperBuild/src/opensfm/opensfm/transformations.py", line 954, in affine_matrix_from_points
u, s, vh = numpy.linalg.svd(numpy.dot(v1, v0.T))
File "/usr/lib/python2.7/dist-packages/numpy/linalg/linalg.py", line 1359, in svd
u, s, vt = gufunc(a, signature=signature, extobj=extobj)
File "/usr/lib/python2.7/dist-packages/numpy/linalg/linalg.py", line 99, in _raise_linalgerror_svd_nonconvergence
raise LinAlgError("SVD did not converge")
numpy.linalg.linalg.LinAlgError: SVD did not converge
Traceback (most recent call last):
File "/code/run.py", line 47, in 
plasm.execute(niter=1)
File "/code/scripts/opensfm.py", line 122, in process
(context.pyopencv_path, context.opensfm_path, tree.opensfm))
File "/code/opendm/system.py", line 34, in run
raise Exception("Child returned {}".format(retcode))
Exception: Child returned 1
--------------------+----------------------------------------------------------

EXIF tags in 'VIRB0001-68.JPG' ('Intel' byte order):
--------------------+----------------------------------------------------------
Tag |Value
--------------------+----------------------------------------------------------
Image Description |DCIM\100_VIRB\VIRB0001-
Manufacturer |Garmin
Model |VIRB Ultra 30
Orientation |Top-left
X-Resolution |72
Y-Resolution |72
Resolution Unit |Inch
Software |4.40
Date and Time |2017:12:29 15:50:38
YCbCr Positioning |Centered
Compression |JPEG compression
X-Resolution |72
Y-Resolution |72
Resolution Unit |Inch
Exposure Time |1/543 sec.
F-Number |f/2.6
Exposure Program |Normal program
ISO Speed Ratings |100
Exif Version |Exif Version 2.3
Date and Time (Origi|2017:12:29 15:50:38
Date and Time (Digit|2017:12:29 15:50:38
Components Configura|- Cr Cb Y
Compressed Bits per |15561472.087
Shutter Speed |9.09 EV (1/543 sec.)
Aperture |2.76 EV (f/2.6)
Exposure Bias |0.00 EV
Maximum Aperture Val|2.76 EV (f/2.6)
Subject Distance |0.0 m
Metering Mode |Center-weighted average
Light Source |Unknown
Flash |No flash function
Focal Length |2.7 mm
Maker Note |256 bytes undefined data
FlashPixVersion |Unknown FlashPix Version
Color Space |sRGB
Pixel X Dimension |4000
Pixel Y Dimension |3000
Exposure Index |0/0
Sensing Method |One-chip color area sensor
File Source |DSC
Scene Type |Directly photographed
Custom Rendered |Normal process
Exposure Mode |Auto exposure
White Balance |Auto white balance
Digital Zoom Ratio | 1
Focal Length in 35mm|15
Scene Capture Type |Landscape
Gain Control |Normal
Contrast |Normal
Saturation |Normal
Sharpness |Normal
Device Setting Descr|4 bytes undefined data
Subject Distance Ran|Unknown
North or South Latit|N
Latitude |40, 32, 56.169895
East or West Longitu|W
Longitude |75, 58, 39.558057
Altitude Reference |Sea level
Altitude |142.470336
Speed Unit |K
Speed of GPS Receive|0.000000
Direction of Movemen|243.863808
Geodetic Survey Data|WGS-84
Interoperability Ind|
Interoperability Ver|0010
--------------------+----------------------------------------------------------
EXIF data contains a thumbnail (3370 bytes).

[image]

dakotabenjamin, 2018-01-04:
I'd like to test this but don't have a Garmin VIRB. Can you find an interesting object and place it on a carpet or other (non-glossy) surface and then take 15-20 photos of it from different angles? Let's try to rule out if it's a camera issue or a scene issue first.

dakotabenjamin, 2018-01-04:
Actually, first lets try to calibrate the camera. It looks like there's some lens distortion. Can you try these scripts: https://github.com/dakotabenjamin/CameraCalibration and test ODM with `--use-fixed-camera-parameters` on the undistorted photos?

bearyjd, 2018-01-04:
I actually stripped the exif data out from all the photos and it worked without any issues.  Once I tried with the same photo set with the exif data back in, odm failed again.

closed 2020-04-14 by smathermather