OpenDroneMap/ODM #735: Use OpenDroneMap for industrial purposes (advanced usage for government applications)

chanakaDe, 2018-01-02:
Hi guys,

I'm Chanaka from Sri Lanka. We have a startup to use drones for industrial mapping. (For government purposes and also map some places hard to access by men). There are 3 basic requirements. 

1. Calculate distance between 2 points.
2. Calculate area of a selected section. (like area of a forest which hard to measure by ground)
3. Height of a pile of sand and rocks (Like small mountains)

Can someone please let me know do we have the ability to perform these 3 tasks using this application? We have a 3DR solo , DJI Phantom 3 standard and also Xiaomi Mi 4K drone. If there are not compatible, we can get any compatible drone. Please advice us on this.

pierotofy, 2018-01-03:
Hey @chanakaDe ✋ 

1. Yes (see WebODM's map interface: https://demo.webodm.org/map/project/3/task/904c1c56-10c5-48df-a104-0138f699dd69/ https://github.com/OpenDroneMap/WebODM )
2. Yes (same as above)
3. Yes (see WebODM's 3D interface https://demo.webodm.org/3d/project/3/task/904c1c56-10c5-48df-a104-0138f699dd69/ )

Drones are fine (I use a DJI Phantom 3 myself).

chanakaDe, 2018-01-04:
@pierotofy  👍  Thanks for the answers buddy. Have one more question to make it clear.
I have Phantom 3 standard version. Can I use it ? Does the images taken from P3 standard have geo locations ? Or do i need to ad them manually using another 3rd party application ? 

And about Xiaomi Mi 4k, it doesn't have geo location i know that for clear. Is there any way I can use it for this application ? Can we add geo location manually via this app ?

pierotofy, 2018-01-04:
Yes you can use ground control points (GCP) if your images don't have GPS coordinates. I think Phantom 3 Standard already has geolocations embedded.