OpenDroneMap/ODM #737: Could not load image header - Cannot determine image format

danbjoseph, 2018-01-05:
ODM installed natively using the master branch on AWS EC2
images in an s3 bucket [mounted](https://cloudkul.com/blog/mounting-s3-bucket-linux-ec2-instance/) using `s3fs`
```
2018-01-04 20:47:54,430 Pruning depthmap for image 20171216_SILVER101_DJI_0530.JPG
2018-01-04 20:51:20,572 Merging depthmaps
[DEBUG]   running PYTHONPATH=/home/ubuntu/OpenDroneMap/SuperBuild/install/lib/python2.7/dist-packages /home/ubuntu/OpenDroneMap/SuperBuild/src/opensfm/bin/export_bundler /home/ubuntu/s3bucket/canaan1/opensfm
[INFO]    Running ODM OpenSfM Cell - Finished
[INFO]    Running ODM Meshing Cell
[DEBUG]   Writing ODM Mesh file in: /home/ubuntu/s3bucket/canaan1/odm_meshing/odm_mesh.ply
[DEBUG]   running /home/ubuntu/OpenDroneMap/build/bin/odm_meshing -inputFile /home/ubuntu/s3bucket/canaan1/opensfm/depthmaps/merged.ply -outputFile /home/ubuntu/s3bucket/canaan1/odm_meshing/odm_mesh.ply -logFile /home/ubuntu/s3bucket/canaan1/odm_meshing/odm_meshing_log.txt -maxVertexCount 100000 -octreeDepth 9  -samplesPerNode 1.0 -solverDivide 9
[INFO]    Running ODM Meshing Cell - Finished
[INFO]    Running MVS Texturing Cell
[DEBUG]   Writing MVS Textured file in: /home/ubuntu/s3bucket/canaan1/odm_texturing/odm_textured_model.obj
[DEBUG]   running /home/ubuntu/OpenDroneMap/SuperBuild/install/bin/texrecon /home/ubuntu/s3bucket/canaan1/opensfm/reconstruction.nvm /home/ubuntu/s3bucket/canaan1/odm_meshing/odm_mesh.ply /home/ubuntu/s3bucket/canaan1/odm_texturing/odm_textured_model -d gmi -o gauss_clamping -t none     
/home/ubuntu/OpenDroneMap/SuperBuild/install/bin/texrecon (built on Jan  3 2018, 19:11:01)
Load and prepare mesh:
PLY Loader: comment PCL generated
Reading PLY: 100053 verts... 199917 faces... done.
Warning: Zero-length normals detected: 4 face normals, 0 vertex normals
Generating texture views:
NVM: Loading file...
NVM: Number of views: 797
NVM: Number of features: 0
        Loading 56%...Could not load image header of /home/ubuntu/s3bucket/canaan1/odm_texturing/tmp/20171216_SILVER101_DJI_0144.JPG.png
/home/ubuntu/s3bucket/canaan1/odm_texturing/tmp/20171216_SILVER101_DJI_0144.JPG.png: Cannot determine image format
Traceback (most recent call last):
  File "/home/ubuntu/OpenDroneMap/run.py", line 47, in <module>
    plasm.execute(niter=1)
  File "/home/ubuntu/OpenDroneMap/scripts/mvstex.py", line 124, in process
    '{keepUnseenFaces}'.format(**kwargs))
  File "/home/ubuntu/OpenDroneMap/opendm/system.py", line 34, in run
    raise Exception("Child returned {}".format(retcode))
Exception: Child returned 1
```

danbjoseph, 2018-01-05:
different error/exception, same place in the process
```
2018-01-04 17:59:52,290 Pruning depthmap for image 20171214_SILVER100_DJI_0638.JPG
2018-01-04 17:59:52,357 Pruning depthmap for image 20171214_MAVIC100_DJI_0664.JPG
2018-01-04 18:04:43,883 Merging depthmaps
[DEBUG]   running PYTHONPATH=/home/ubuntu/OpenDroneMap/SuperBuild/install/lib/python2.7/dist-packages /home/ubuntu/OpenDroneMap/SuperBuild/src/opensfm/bin/export_bundler /home/ubuntu/s3bucket/canaan0/opensfm
[INFO]    Running ODM OpenSfM Cell - Finished
[INFO]    Running ODM Meshing Cell
[DEBUG]   Writing ODM Mesh file in: /home/ubuntu/s3bucket/canaan0/odm_meshing/odm_mesh.ply
[DEBUG]   running /home/ubuntu/OpenDroneMap/build/bin/odm_meshing -inputFile /home/ubuntu/s3bucket/canaan0/opensfm/depthmaps/merged.ply -outputFile /home/ubuntu/s3bucket/canaan0/odm_meshing/odm_mesh.ply -logFile /home/ubuntu/s3bucket/canaan0/odm_meshing/odm_meshing_log.txt -maxVertexCount 100000 -octreeDepth 9  -samplesPerNode 1.0 -solverDivide 9
[INFO]    Running ODM Meshing Cell - Finished
[INFO]    Running MVS Texturing Cell
[DEBUG]   Writing MVS Textured file in: /home/ubuntu/s3bucket/canaan0/odm_texturing/odm_textured_model.obj
[DEBUG]   running /home/ubuntu/OpenDroneMap/SuperBuild/install/bin/texrecon /home/ubuntu/s3bucket/canaan0/opensfm/reconstruction.nvm /home/ubuntu/s3bucket/canaan0/odm_meshing/odm_mesh.ply /home/ubuntu/s3bucket/canaan0/odm_texturing/odm_textured_model -d gmi -o gauss_clamping -t none     
/home/ubuntu/OpenDroneMap/SuperBuild/install/bin/texrecon (built on Jan  3 2018, 19:20:27)
Load and prepare mesh:
PLY Loader: comment PCL generated
Reading PLY: 100061 verts... 199920 faces... done.
Generating texture views:
NVM: Loading file...
NVM: Number of views: 726
NVM: Number of features: 0
        Loading 10%...terminate called after throwing an instance of 'util::Exception'
  what():  /home/ubuntu/s3bucket/canaan0/opensfm/undistorted/20171214_MAVIC100_DJI_0979.JPG.jpg: Cannot determine image format
Aborted (core dumped)
Traceback (most recent call last):
  File "/home/ubuntu/OpenDroneMap/run.py", line 47, in <module>
    plasm.execute(niter=1)
  File "/home/ubuntu/OpenDroneMap/scripts/mvstex.py", line 124, in process
    '{keepUnseenFaces}'.format(**kwargs))
  File "/home/ubuntu/OpenDroneMap/opendm/system.py", line 34, in run
    raise Exception("Child returned {}".format(retcode))
Exception: Child returned 134
```

dakotabenjamin, 2018-01-08:
Does /home/ubuntu/s3bucket/canaan1/odm_texturing/tmp/20171216_SILVER101_DJI_0144.JPG.png
/home/ubuntu/s3bucket/canaan1/odm_texturing/tmp/20171216_SILVER101_DJI_0144.JPG.png still exist?

danbjoseph, 2018-01-08:
pretty sure i terminated that particular instance a while back.
haven't run into the same issue if i copy the files over first.
can try and reproduce if you're curious.

dakotabenjamin, 2018-03-28:
Closing this- reopen if the issue happens again

closed 2018-03-28 by dakotabenjamin