OpenDroneMap/ODM #739: Aborted (core dumped)

danbjoseph, 2018-01-09:
[log-61.txt](https://github.com/OpenDroneMap/OpenDroneMap/files/1613699/log-61.txt)

```
[92m[DEBUG]   running /home/ubuntu/OpenDroneMap/SuperBuild/install/bin/texrecon /home/ubuntu/project/canaan61/opensfm/reconstruction.nvm /home/ubuntu/project/canaan61/odm_meshing/odm_mesh.ply /home/ubuntu/project/canaan61/odm_texturing/odm_textured_model -d gmi -o gauss_clamping -t none     [0m
/home/ubuntu/OpenDroneMap/SuperBuild/install/bin/texrecon (built on Jan  5 2018, 13:25:57)
Load and prepare mesh: 
PLY Loader: comment PCL generated
Reading PLY: 100019 verts... 199876 faces... done.
Generating texture views: 
NVM: Loading file...
NVM: Number of views: 718
NVM: Number of features: 0

                                                                                
	Loading 100%... done. (Took 82.536s)
Building adjacency graph: 

                                                                                
	Adding edges 100%... done. (Took 0.401s)
	299735 total edges.
View selection:
	Building BVH from 199876 faces... done. (Took: 142 ms)

                                                                                
	Calculating face qualities 100%... done. (Took 51.128s)

                                                                                
	Postprocessing face infos 100%... done. (Took 0.466s)
	Maximum quality of a face within an image: 14765.4
	Clamping qualities to 1453.06 within normalization.
	Writing data cost file... done.
	Optimizing:
		Time[s]	Energy
		0	157477
Aborted (core dumped)
Traceback (most recent call last):
  File "/home/ubuntu/OpenDroneMap/run.py", line 47, in <module>
    plasm.execute(niter=1)
  File "/home/ubuntu/OpenDroneMap/scripts/mvstex.py", line 124, in process
    '{keepUnseenFaces}'.format(**kwargs))
  File "/home/ubuntu/OpenDroneMap/opendm/system.py", line 34, in run
    raise Exception("Child returned {}".format(retcode))
Exception: Child returned 134
```

danbjoseph, 2018-01-09:
rebooted the instance and tried to run it again
[log-61_2.txt](https://github.com/OpenDroneMap/OpenDroneMap/files/1615196/log-61_2.txt)

```
^[[94m[INFO]    Found 718 usable images^[[0m
^[[94m[INFO]    Running ODM Load Dataset Cell - Finished^[[0m
^[[94m[INFO]    Running ODM OpenSfM Cell^[[0m
^[[93m[WARNING] Found a valid OpenSfM reconstruction file in: /home/ubuntu/project/canaan61/opensfm/reconstruction.json^[[0m
^[[93m[WARNING] Found a valid Bundler file in: /home/ubuntu/project/canaan61/opensfm/reconstruction.json^[[0m
^[[94m[INFO]    Running ODM OpenSfM Cell - Finished^[[0m
^[[94m[INFO]    Running ODM Meshing Cell^[[0m
^[[93m[WARNING] Found a valid ODM Mesh file in: /home/ubuntu/project/canaan61/odm_meshing/odm_mesh.ply^[[0m
^[[94m[INFO]    Running ODM Meshing Cell - Finished^[[0m
^[[94m[INFO]    Running MVS Texturing Cell^[[0m
^[[92m[DEBUG]   Writing MVS Textured file in: /home/ubuntu/project/canaan61/odm_texturing/odm_textured_model.obj^[[0m
^[[92m[DEBUG]   running /home/ubuntu/OpenDroneMap/SuperBuild/install/bin/texrecon /home/ubuntu/project/canaan61/opensfm/reconstruction.nvm /home/ubuntu/project/canaan61/odm_meshing/odm_mesh.ply /home/ubuntu/project/canaan61/odm_texturing/odm_textured_model -d gmi -o gauss_clamping -t none     ^[[0m
/home/ubuntu/OpenDroneMap/SuperBuild/install/bin/texrecon (built on Jan  5 2018, 13:25:57)
Temporary directory "tmp" exists within the destination directory.
Cannot continue since this directory would be delete in the end.

Traceback (most recent call last):
  File "/home/ubuntu/OpenDroneMap/run.py", line 47, in <module>
    plasm.execute(niter=1)
  File "/home/ubuntu/OpenDroneMap/scripts/mvstex.py", line 124, in process
    '{keepUnseenFaces}'.format(**kwargs))
  File "/home/ubuntu/OpenDroneMap/opendm/system.py", line 34, in run
    raise Exception("Child returned {}".format(retcode))
Exception: Child returned 1
```

dakotabenjamin, 2018-01-09:
Can you delete the odm_texturing folder and try again?

danbjoseph, 2018-01-09:
too late, it's all gone. the world may never know.
:\

danbjoseph, 2018-01-09:
so... running the same dataset on a different instance. first run hit what looks like the same issue from the first run on the original instance. i'll wait to do anything this time until i hear from you about what might help troubleshoot this.
```
^[[94m[INFO]    Running ODM Meshing Cell - Finished^[[0m
^[[94m[INFO]    Running MVS Texturing Cell^[[0m
^[[92m[DEBUG]   Writing MVS Textured file in: /home/ubuntu/project/canaan61/odm_texturing/odm_textured_model.obj^[[0m
^[[92m[DEBUG]   running /home/ubuntu/OpenDroneMap/SuperBuild/install/bin/texrecon /home/ubuntu/project/canaan61/opensfm/reconstruction.nvm /home/ubuntu/project/canaan61/odm_meshing/odm_mesh.ply /home/ubuntu/project/canaan61/odm_texturing/odm_textured_model -d gmi -o gauss_clamping -t none     ^[[0m
/home/ubuntu/OpenDroneMap/SuperBuild/install/bin/texrecon (built on Jan  5 2018, 13:25:57)
Load and prepare mesh:
PLY Loader: comment PCL generated
Reading PLY: 100020 verts... 199875 faces... done.
Warning: Zero-length normals detected: 4 face normals, 0 vertex normals
Generating texture views:
NVM: Loading file...
NVM: Number of views: 718
NVM: Number of features: 0
^M                                                                                ^M    Loading 100%... done. (Took 82.019s)
Building adjacency graph:
^M                                                                                ^M    Adding edges 100%... done. (Took 0.399s)
        299731 total edges.
View selection:
        Building BVH from 199875 faces... done. (Took: 237 ms)
^M                                                                                ^M    Calculating face qualities 100%... done. (Took 50.891s)
^M                                                                                ^M    Postprocessing face infos 100%... done. (Took 0.466s)
        Maximum quality of a face within an image: 14131.6
        Clamping qualities to 1468.42 within normalization.
        Writing data cost file... done.
        Optimizing:
                Time[s] Energy
                0       158319
Aborted (core dumped)
Traceback (most recent call last):
  File "/home/ubuntu/OpenDroneMap/run.py", line 47, in <module>
    plasm.execute(niter=1)
  File "/home/ubuntu/OpenDroneMap/scripts/mvstex.py", line 124, in process
    '{keepUnseenFaces}'.format(**kwargs))
  File "/home/ubuntu/OpenDroneMap/opendm/system.py", line 34, in run
    raise Exception("Child returned {}".format(retcode))
Exception: Child returned 134
```

r2d3, 2018-01-16:
Hi @danbjoseph @dakotabenjamin 

I have a similar crash in texrecon from ODM 0.3 ... I recompiled the latest Git version and the problem disappeared. Recompiled with clang-3.5 + AddressSanitizer and stock g++ 4.8.4

So the problem has been corrected in some way ?!

pierotofy, 2018-01-22:
Related: http://community.opendronemap.org/t/best-options-for-orthoimage/175/9?u=pierotofy

dakotabenjamin, 2018-03-28:
This issue was moved to http://community.opendronemap.org/t/aborted-core-dumped/387

closed 2018-03-28 by dakotabenjamin