OpenDroneMap/ODM #741: merged depthmaps failure

joemcglinchy, 2018-01-10:
Hello! first of all, big thanks to all of you who are working on this project. This is awesome, and I'm hopeful for using this software for photogrammetry projects. 

I had done some looking around on this forum to see about better processing for a forested area, and it seems like the --use-25dmesh is the way to go for that. Is this correct? I had been trying with PMVS and seeing the famous 'blobs' in areas of tree canopy. Here is an error I am receiving:

```...
2018-01-10 22:16:43,194 Pruning depthmap for image IMG_170921_192903_0044__refl_RED.jpg
2018-01-10 22:16:43,944 Pruning depthmap for image IMG_170921_192807_0022__refl_RED.jpg
2018-01-10 22:16:43,999 Pruning depthmap for image IMG_170921_192945_0059__refl_RED.jpg
2018-01-10 22:16:44,907 Merging depthmaps
Killed
Traceback (most recent call last):
  File "/code/run.py", line 47, in <module>
    plasm.execute(niter=1)
  File "/code/scripts/opensfm.py", line 145, in process
    (context.pyopencv_path, context.opensfm_path, tree.opensfm))
  File "/code/opendm/system.py", line 34, in run
    raise Exception("Child returned {}".format(retcode))
Exception: Child returned 137
```

I am using the docker implementation with these commands:
```docker run -it --rm -v C:\Projects\Forests\uav\reflectance_proc_ODM_2\images\red:/code/images ^
-v C:\Projects\Forests\uav\reflectance_proc_ODM_2\images\odm_orthophoto:/code/odm_orthophoto ^
-v C:\Projects\Forests\uav\reflectance_proc_ODM_2\images\odm_georeferencing:/code/odm_georeferencing ^
-v C:\Projects\Forests\uav\reflectance_proc_ODM_2\images\odm_dem:/code/odm_dem ^
-v C:\Projects\Forests\uav\reflectance_proc_ODM_2\images\odm_dsm:/code/odm_dsm ^
opendronemap/opendronemap ^
--mesh-size 100000 ^
--min-num-features 10000 ^
--force-focal 4.0 ^
--force-ccd 226.666 ^
--use-25dmesh ^
--dtm ^
--dsm
```

Any thoughts on the reason for this failure?

Jakst3r, 2018-01-11:
I'm getting the same error.

joemcglinchy, 2018-01-11:
Thanks @Jakst3r , that makes sense I guess. I only saw this when adding in the DTM/DSM output. I increased the memory in my Docker (windows) settings and with that same command I had everything run successfully, even if the DSM production took a while.

As another comment, and please point me to the correct thread if this is not appropriate here, we are collecting data over a forested area which is in recovery, so definitely not a dense forested area. Using the 2.5d mesh flag to produce the orthophotos still results in some blobby areas where there are tree stems. I'd like to play with some settings to see if I can get a better output, but is that possible using Docker to run ODM and pick up at some stage in the processing so as to not run through the entire pipeline every time?

dakotabenjamin, 2018-01-11:
Add -v /<path>/opensfm/:/code/opensfm as a volume, this is where the bulk of operations happen. If there's already a model in there it won't rerun unless you specify with --rerun parameters.

joemcglinchy, 2018-01-11:
would I then have to run the docker command without --rm? If I specify the container to be removed at the end of execution then that data will not be there, correct?

dakotabenjamin, 2018-01-12:
You can, just make sure you remove the container when you're done.

closed 2018-03-17 by dakotabenjamin