OpenDroneMap/ODM #744: libpng-Error, Child returned 134

useher, 2018-01-17:
Hello!
There is an error occuring  while/after merging depthmaps. When i run
`docker run -it --rm -v $(pwd)/data/image:/code/images -v $(pwd)/data/odm_georeferencing:/code/odm_georeferencing -v $(pwd)/data/odm_orthophote:/code/odm_orthophoto opendronemap/opendronemap

` like in the documentation i get the following error:

`
[...]
2018-01-17 10:09:16,897 Pruning depthmap for image DJI_0560.JPG
2018-01-17 10:09:19,621 Merging depthmaps
[DEBUG]   running PYTHONPATH=/code/SuperBuild/install/lib/python2.7/dist-packages /code/SuperBuild/src/opensfm/bin/export_bundler /code/opensfm
[INFO]    Running ODM OpenSfM Cell - Finished
[INFO]    Running ODM Meshing Cell
[DEBUG]   Writing ODM Mesh file in: /code/odm_meshing/odm_mesh.ply
[DEBUG]   running /code/build/bin/odm_meshing -inputFile /code/opensfm/depthmaps/merged.ply -outputFile /code/odm_meshing/odm_mesh.ply -logFile /code/odm_meshing/odm_meshing_log.txt -maxVertexCount 100000 -octreeDepth 9  -samplesPerNode 1.0 -solverDivide 9
[INFO]    Running ODM Meshing Cell - Finished
[INFO]    Running MVS Texturing Cell
[DEBUG]   Writing MVS Textured file in: /code/odm_texturing/odm_textured_model.obj
[DEBUG]   running /code/SuperBuild/install/bin/texrecon /code/opensfm/reconstruction.nvm /code/odm_meshing/odm_mesh.ply /code/odm_texturing/odm_textured_model -d gmi -o gauss_clamping -t none     
/code/SuperBuild/install/bin/texrecon (built on Jan 16 2018, 00:27:05)
Load and prepare mesh: 
PLY Loader: comment PCL generated
Reading PLY: 51307 verts... 102460 faces... done.
Warning: Zero-length normals detected: 48 face normals, 0 vertex normals
Generating texture views: 
NVM: Loading file...
NVM: Number of views: 349
NVM: Number of features: 0
        Loading 74%...libpng error: Write Error                                 
libpng error: Write Error
Aborted (core dumped)
Traceback (most recent call last):
  File "/code/run.py", line 47, in <module>
    plasm.execute(niter=1)
  File "/code/scripts/mvstex.py", line 124, in process
    '{keepUnseenFaces}'.format(**kwargs))
  File "/code/opendm/system.py", line 34, in run
    raise Exception("Child returned {}".format(retcode))
Exception: Child returned 134`

There is no difference in building the container by myself or using the syntax above. 
libpng is Version 1.6, the devel-package is installed, the underlying system is openSUSE Leap 42.2 with 8 cores and 16GB RAM. I try to create an orthophoto from 407 images with around 2GB all together. Can you give me a hint what to do to handle this? 
Thank you in advance and keep up the good work!
Uwe Seher

dakotabenjamin, 2018-01-17:
THat error usually occurs when you run out of hard drive space. Can you check that?

useher, 2018-01-18:
235G free should be enough, i think. I will try if later on another machine with more RAM and report here.

useher, 2018-01-19:
I started odm on our server with lots of cores and ram and it worked good, so i think this should be a problem with resources. As a sideeffect, the apc started beeping immediately after the process started, so i think, i have to do some finetuning here.

closed 2018-03-28 by dakotabenjamin

amayler, 2019-01-14:
I have the same problem and I can't find a solution to it. 
I am testing a set of 13 jpg images of 60M each. I run the native odm in ubuntu 16.04, with 64G RAM, 32 nuclei and a lot of hard disk space (700G approx). But error 134 still appears:

	Took: 65.464s
Generating texture patches:
	Running... done. (Took 10.034s)
	113 texture patches.
Running global seam leveling:
	Create matrices for optimization...  done.
	Lhs dimensionality: 93817 x 93817
	Calculating adjustments:
		Color channel 0: CG took 66 iterations. Residual is 9.0549e-05
		Color channel 1: CG took 66 iterations. Residual is 9.0549e-05
		Color channel 2: CG took 66 iterations. Residual is 9.0549e-05
		Took 0.233 seconds
        Adjusting texture patches 100%... done. (Took 13.912s)                  
Running local seam leveling:
        Blending texture patches 100%... done. (Took 33.412s)                   
Generating texture atlases:
	Sorting texture patches... done.
	Working on atlas 565 90%... terminate called after throwing an instance of 'std::bad_alloc'
  what():  std::bad_alloc
Aborted (core dumped)
Traceback (most recent call last):
  File "run.py", line 47, in <module>
    plasm.execute(niter=1)
  File "/home/inaoe01/ODM/scripts/mvstex.py", line 126, in process
    '-n {nadirWeight}'.format(**kwargs))
  File "/home/inaoe01/ODM/opendm/system.py", line 34, in run
    raise Exception("Child returned {}".format(retcode))
Exception: Child returned 134

I understand that the problem is the size of the images (since reducing their size, works) but in this case the quality of the images is very important and can not be reduced. I thought that with a computer with good computing resources the problem would be solved, but it is not so.  

Any help? Thank you in advance.