OpenDroneMap/ODM #746: `alt: None` results in match_candidates error?

danbjoseph, 2018-01-18:
[log.txt](https://github.com/OpenDroneMap/OpenDroneMap/files/1643719/log.txt)

```
Loaded canaan61_orthophoto.tif.jpg | camera:  | dimensions: 39659 x 26483 | focal: None | ccd: 5.75 | lat: 2066122.489 | lon: -784996.973 | alt: None
```
```
2018-01-18 14:32:13,220 Found 5365 points in 1.63809704781s
2018-01-18 14:32:14,846 Found 5449 points in 1.64672994614s
2018-01-18 14:32:26,054 Found 5544 points in 1.56275320053s
^[[92m[DEBUG]   running PYTHONPATH=/home/ubuntu/OpenDroneMap/SuperBuild/install/lib/python2.7/dist-packages /home/ubuntu/OpenDroneMap/SuperBuild/src/opensfm/bin/opensfm match_features /home/ubuntu/projects/canaan/opensfm^[[0m
Traceback (most recent call last):
  File "/home/ubuntu/OpenDroneMap/SuperBuild/src/opensfm/bin/opensfm", line 34, in <module>
    command.run(args)
  File "/home/ubuntu/OpenDroneMap/SuperBuild/src/opensfm/opensfm/commands/match_features.py", line 29, in run
    pairs = match_candidates_from_metadata(images, exifs, data)
  File "/home/ubuntu/OpenDroneMap/SuperBuild/src/opensfm/opensfm/commands/match_features.py", line 167, in match_candidates_from_metadata
    gps_neighbors, max_distance)
  File "/home/ubuntu/OpenDroneMap/SuperBuild/src/opensfm/opensfm/commands/match_features.py", line 90, in match_candidates_by_distance
    gps['latitude'], gps['longitude'], gps['altitude'],
KeyError: 'altitude'
Traceback (most recent call last):
  File "/home/ubuntu/OpenDroneMap/run.py", line 47, in <module>
    plasm.execute(niter=1)
  File "/home/ubuntu/OpenDroneMap/scripts/opensfm.py", line 106, in process
    (context.pyopencv_path, context.opensfm_path, tree.opensfm))
  File "/home/ubuntu/OpenDroneMap/opendm/system.py", line 34, in run
    raise Exception("Child returned {}".format(retcode))
Exception: Child returned 1
```

dakotabenjamin, 2018-03-09:
Sorry for the late response :pray:. It's probably an opensfm issue if anything. I'm not sure opensfm is equipped to handle merged orthos.

smathermather, 2018-03-16:
I think the conclusion was there wasn't enough overlap between the orthos for this approach. I will close for now. Please re-open at will!

closed 2018-03-16 by smathermather