OpenDroneMap/ODM #748: Export undistorted images in a lossless format for texturing

pierotofy, 2018-01-21:
There's a bit of background about this problem here: https://github.com/nmoehrle/mvs-texturing/issues/90

In short, when images are undistorted in jpg format, the border of images is not entirely black and  regions that should be invalid might be classified as valid and used in the reconstruction.

This causes artifacts to appear in the orthophotos in form of small lines.

The proposed fix would involve exporting the images in PNG instead of JPG.

juvinski, 2018-01-21:
Hi @pierotofy , how can I help ?

best regards

pierotofy, 2018-01-21:
@juvinski if you know some Python I think the first step would be to add a flag to the opensfm's `undistort` command to select the type of file output (jpg or png): https://github.com/mapillary/OpenSfM/blob/59efee3a7043a441aa64c013dd7fff611d39194c/opensfm/dataset.py#L85
https://github.com/mapillary/OpenSfM/blob/684ebff2ecfc2017b64fcfc18fc1fdbe14811925/opensfm/commands/undistort.py#L95

pierotofy, 2019-05-23:
This has been implemented some time ago. We now use PNGs for texturing.

closed 2019-05-23 by pierotofy