OpenDroneMap/ODM #750: Asphalt and roads/pathways in general never present at points cloud

cleberzavadniak, 2018-01-22:
Hi!

I work at DroneMapp, a Brazilian company that offers a nice platform to manage geo-information, and we plan to use OpenDroneMap there as an option to our customers to process their images, usually taken from drones.

I really like the results of OpenDroneMap, and I even wrote an article (in Portuguese) comparing some of them with Pix4D, and I must say OpenDroneMap did very well in general, even comparing with a proprietary platform (with lots of budget...).

https://medium.com/dronemapp/lan%C3%A7ada-a-plataforma-dronemapp-para-pilotos-de-drones-6d995a832238
(Non-Portuguese-speaking people can look at the pictures, at least, and **see** what I'm talking about.)

Well, right now we have only one major issue: although roads in general look just fine at the orthomosaics, like this,

[image]

they never show up at the points clouds, like this,

[image]
.

That sometimes leads to very weird renderings of the orthomosaic itself, since it is based on the points cloud (right?), like this:

[image]

Since we plan to sell our services to road building companies, we'd be going to have a hard time using OpenDroneMap.

Is there something I could do to improve the detection of asphalt, at least? Should I try to tweak OpenSFM or something like that?

dakotabenjamin, 2018-01-22:
Google translate did a pretty good job for me 👍  I'm happy to see positive press about ODM! Thanks! I'd like to know, what version of ODM were you using to compare with pix4d? 

Some things I would try for detecting asphalt (using the master branch):
 - skip resizing the images: `--resize-to 0`
 - increase feature matching: `--min-num-features 10000`
 
To improve the bridge (and other complex topologies where you are not concerned about the 3D product as much):
- Try using the 2.5d meshing module: `--use-25d-mesh`

cleberzavadniak, 2018-01-22:
:+1: 

Right now, I am using the docker image, tag ":wb" (see #747). I'm going to try `latest` tomorrow.

Actually, I already disabled the image resizing. (The command line argument should be `--resize-to -1'.)

I'm going to try both other options (increase feature matching and using 2.5D mesh) soon.

Increasing the minimum number of features makes the matching process go on until that number is reached? Or is there a risk, maybe,that some images could be discarded?

Anyway, thanks for answering. I will post the results here after the processing is finished.

dakotabenjamin, 2018-01-22:
It increases the size of the sparse point cloud, which I'm hoping will catch more points in the asphalt. I think now that maybe you need to adjust the parameters for the dense point cloud instead. That will take some code tweaking. 

Here are the parameters you can adjust for depth map generation: https://github.com/mapillary/OpenSfM/blob/master/opensfm/config.py#L113

We have stuck to the default parameters because I'm not exactly sure how they will affect the model. You can adjust this in ODM by adding them to the code here: https://github.com/OpenDroneMap/OpenDroneMap/blob/master/scripts/opensfm.py#L72

For example, to change the depth map resolution (and increase the size of the point cloud), you would add an element to the `config` list:
```
    "depthmap_resolution: 800",
```

cleberzavadniak, 2018-01-26:
I'm back with some news.

First: `--min-num-features 10000` is **way. too. slow**, so I gave up after almost 20 hours of processing and tried `--min-num-features 5000`, together with `--use-25dmesh`.

The resulting points cloud:

[image]

It is already a little better than the original one, since I see a little more points of asphalt.
(I don't have the orthomosaic file available right now, sorry.)

Next I'm going to try with `--min-num-features 7500` and no 2.5D mesh.

I plan to tweak opensfm only as last resort...

dakotabenjamin, 2018-01-26:
I was tweaking the opensfm parameters earlier and unfortunately even though the number of points increase (which was good), it wasn't in areas where I wanted them, like roads. This may be an issue that should be brought up with mapillary over at https://github.com/mapillary/OpenSfM

cleberzavadniak, 2018-01-26:
Could you hand me some examples of what parameters you tried? It seems fit to open a similar Issue at OpenSfM repo, indeed, so I want to provide the most useful content I can.

smathermather, 2018-02-11:
@cleberzavadniak -- these points are being thrown out because of the SD filtering on the patches. Change your https://github.com/mapillary/OpenSfM/blob/master/opensfm/config.py#L118 to a lower value to get these filled back in. You may also want to increase https://github.com/mapillary/OpenSfM/blob/master/opensfm/config.py#L121 to 4 or 5 to compensate for the additional noise in the point cloud.

smathermather, 2018-02-13:
@cleberzavadniak -- did you get a chance to try this? I have found it works quite well in our tests.

cleberzavadniak, 2018-02-14:
I'm going "back to lab" just now. I'm going to try again and then I post here the results. Sorry for the delay.

pierotofy, 2018-02-24:
See also an analysis of opensfm's depthmaps parameters here: #662

smathermather, 2018-11-22:
This is fixed somewhere around this commit: https://github.com/OpenDroneMap/ODM/commit/e99df871f71528bf459135e22ea4659419efcb6c. Closing!

closed 2018-11-22 by smathermather