OpenDroneMap/ODM #751: Illegal Instruction.... Exception: Child returned 132

neosaadmin, 2018-01-25:
OpenDroneMap installed Native on a server AMD 6172 x4: 48 Cores; 64gb RAM; 200Gb space. Ubuntu 16.04LTS Server.

If I attempt to run ODM via CLI or via node-opendronemap (Native install) I get the following:
(./run.sh project)

NVM: Number of features: 0
Illegal instruction
Traceback (most recent call last):
  File "/home/mjpotter/OpenDroneMap/run.py", line 46, in <module>
    plasm.execute(niter=1)
  File "/home/mjpotter/OpenDroneMap/scripts/mvstex.py", line 109, in process
    '{keepUnseenFaces}'.format(**kwargs))
  File "/home/mjpotter/OpenDroneMap/opendm/system.py", line 28, in run
    raise Exception("Child returned {}".format(retcode))
Exception: Child returned 132

I used data sets from inside the ODM sets; also ran the same set in a WebODM virtualbox on my laptop that worked without issue.

Is there something that I am missing?

pierotofy, 2018-01-26:
Same as https://github.com/OpenDroneMap/WebODM/issues/378

dakotabenjamin, 2018-01-26:
I have this issue as well. 

There seems to be a problem with how mvs_texturing is doing temporary files.

neosaadmin, 2018-01-26:
I did, but I will `./configure.sh reinstall`

neosaadmin, 2018-01-26:
Attempted reinstall, and delete OpenDroneMap and `./configure.sh install` to no avail.

sreynolds017, 2018-04-12:
I encountered the same "Exception: Child returned 132" issue at the same point in the processing workflow as described above.  However, my experience was on a stand-alone native install of ODM, rather than a server install as described by the OP.  I don't know if that might make a difference with respect to the ultimate solution, but I figured that I'd mention it just in case it does.

ramunasd, 2018-10-29:
Same for me. Using pre and manually built docker container.
Looks like OpenSfM use some fancy asembler instructions which are not available on older CPU.

closed 2020-04-14 by smathermather