OpenDroneMap/ODM #752: Support for crop height calculation (DSM - DTM)

pierotofy, 2018-01-26:
This is present in the `lidar2dems` code, should be fairly simple to implement by subtracting DTM from DSM. It was suggested on gitter.

yjmenezes, 2018-01-26:
@pierotofy 

Point classification from RGB photos is a hard task.
I plan try lidar2dems.  http://applied-geosolutions.github.io/lidar2dems/

I upload to my git a small urban area.  10 photos + LiDAR.
I have rural areas too. Some flight.

I hope it can be usefull for tests.

bipulneupane, 2018-01-30:
@pierotofy thank you for your help and apologies for late gratitude.

@yjmenezes Thank you for your help. Can i make CHM out of the DSM and DTM images that are created not by using l2d? I want to get CHM from the outputs of WebODM? Is there any limitation while doing so?

yjmenezes, 2018-01-30:
Hi @bipulneupane 

As you, i am looking for .

I am note sure if @hobu  PDAL has a filter to produce bare earth DTM from ODM point cloud (XYZ+RGB).
https://www.pdal.io/stages/filters.html

The problem is shown here:
https://rapidlasso.com/2015/09/21/creating-dtms-from-dense-matched-points-of-uav-imagery-from-senseflys-ebee/

I upload a LiDAR + RGB data set in a hope that  someone  could use it  and help to make ODM even better.  ODM is a great work. Thank you ODM developers.

bipulneupane, 2018-02-08:
@yjmenezes 
Hi!

I wanted to know 
1. if it is always necessary to classify the point cloud first
2. Error while density:
    $ l2d_dems density georeferenced_model.las 
    > Error creating density : No LAS files found
    > l2d_dems density completed (.) in 0:00:00.000208

I am getting error on the first step of classifying.

yjmenezes, 2018-02-08:
@bipulneupane 

I think you must point the directory and not a las file.
l2d_dems density  /path_to_las/

i have never used l2d_dems, i am sorry.
I think it's primary use 
[image]
is for LiDAR data.

btw, I upload a LAS LiDAR dataset . It is not carefully classified but has all laser return pulses.
Class (2)  would be bare earth.
https://github.com/yjmenezes/church_dataset

GeospatialDaryl, 2018-02-08:
LiDAR data (like in the referenced diagram) includes an intensity value;
which is the response peaks on the graph.  SfM PC will not have that
information.

To develop a CSM, you will need to develop a Ground Surface model, and then
use that to classify your point cloud, then calculate separation of those
surfaces.

dakotabenjamin, 2018-03-17:
I can see this being implemented as a contrib/ module

lucadelu, 2018-04-13:
[Here](https://gitlab.com/trilogis/stem/blob/master/libs/las_stem.py#L428)  you can find Python code using PDAL (probably it should be updated) to calculate CHM from input LAS file and DTM.

Do you think could be useful? I could help to adapt to OpenDroneMap with some suggestions

lucadelu, 2018-04-16:
@pierotofy thanks for your answer, yes we should use to use JSON, this was old code..

About CHM calculation we could have several way to calculate:
* from existing rasters but at this point we could do this using existing software like GRASS GIS through mapcalc operation
* from an existing DTM and the calculated DSM from OpenDroneMap (still using GRASS GIS)
* from  an existing DTM and the OpenDroneMap point cloud (using PDAL)

pierotofy, 2021-12-07:
No bandwidth to implement this, although a contrib module would be a most welcome contribution.

closed 2021-12-07 by pierotofy