OpenDroneMap/ODM #754: enabling fast-orthophoto causes 25dmeshing seg fault

PeterSprague, 2018-01-31:
Successfully runs and produces ortho w/o fast-orthophoto enabled.  Opensfm_processes set to 12 on a 32 core 64Gb workstation.

With fast-orthophoto enabled:
2018-01-31 20:43:33,473 DEBUG: Undistorting image 2017_0909_114819_028.JPG
2018-01-31 20:43:33,505 DEBUG: Undistorting image 2017_0909_115521_230.JPG
[DEBUG]   running PYTHONPATH=/code/SuperBuild/install/lib/python2.7/dist-packages /code/SuperBuild/src/opensfm/bin/opensfm export_ply --no-cameras /var/www/data/60dc1e42-05d0-4b17-af63-e03077f64b56/opensfm
[DEBUG]   running PYTHONPATH=/code/SuperBuild/install/lib/python2.7/dist-packages /code/SuperBuild/src/opensfm/bin/export_bundler /var/www/data/60dc1e42-05d0-4b17-af63-e03077f64b56/opensfm
[INFO]    Running ODM OpenSfM Cell - Finished
[INFO]    Running ODM Meshing Cell
[DEBUG]   Writing ODM 2.5D Mesh file in: /var/www/data/60dc1e42-05d0-4b17-af63-e03077f64b56/odm_meshing/odm_25dmesh.ply
[DEBUG]   running /code/build/bin/odm_25dmeshing -inputFile /var/www/data/60dc1e42-05d0-4b17-af63-e03077f64b56/opensfm/reconstruction.ply -outputFile /var/www/data/60dc1e42-05d0-4b17-af63-e03077f64b56/odm_meshing/odm_25dmesh.ply -logFile /var/www/data/60dc1e42-05d0-4b17-af63-e03077f64b56/odm_meshing/odm_25dmeshing_log.txt -maxVertexCount 30000 -neighbors 24 -resolution 0 -verbose
Loading point cloud... loaded 167044 points
Remove outliers... 8878 points removed
Squash point cloud to plane... Segmentation fault (core dumped)
Traceback (most recent call last):
  File "/code/run.py", line 47, in <module>
    plasm.execute(niter=1)
  File "/code/scripts/odm_meshing.py", line 105, in process
    '-resolution {resolution} {verbose}'.format(**kwargs))
  File "/code/opendm/system.py", line 34, in run
    raise Exception("Child returned {}".format(retcode))
Exception: Child returned 139

pierotofy, 2018-01-31:
Could you share your `/var/www/data/60dc1e42-05d0-4b17-af63-e03077f64b56/opensfm/reconstruction.ply`? This might be a bug.

Thank you!

PeterSprague, 2018-01-31:
Here is the file.  I am using VTK v8.1.0 built from source, not v7.1.1 as originally specified in cmake
[reconstruction.ply.txt](https://github.com/OpenDroneMap/OpenDroneMap/files/1683385/reconstruction.ply.txt)

pierotofy, 2018-01-31:
That might be the root cause of the issue. I ran this on a VTK7 build and the task ran successfully. 

```
./odm_25dmeshing -inputFile ./reconstruction.ply -outputFile test_output.ply -verbose
Loading point cloud... loaded 167044 points
Remove outliers... 8878 points removed
Squash point cloud to plane... OK
Automatically set resolution to 3.962223
Plane extentX: 280.582718, extentY: 403.588531
DSM size is 1112x1600 (8.000000 MB) 
Point interpolation using shepard's kernel... OK
Transfering interpolation results to DSM... OK
Median filter...OK
Triangulate... OK
Transform... OK
Saving mesh to file... OK
```

[test_output.ply.txt](https://github.com/OpenDroneMap/OpenDroneMap/files/1683402/test_output.ply.txt)

PeterSprague, 2018-01-31:
Check the errors here: https://github.com/mapillary/OpenSfM/issues/277
Almost exactly the same.  Would your core2 solution work here?

PeterSprague, 2018-02-01:
Reverted to VTK v7.1.1 built from source, includes vtk6-dev pkg  opensf_processes @16

[INFO]    Running ODM OpenSfM Cell - Finished
[INFO]    Running ODM Meshing Cell
[DEBUG]   Writing ODM 2.5D Mesh file in: /var/www/data/34d695f9-4899-45d7-bb8a-21d369d8623f/odm_meshing/odm_25dmesh.ply
[DEBUG]   running /code/build/bin/odm_25dmeshing -inputFile /var/www/data/34d695f9-4899-45d7-bb8a-21d369d8623f/opensfm/reconstruction.ply -outputFile /var/www/data/34d695f9-4899-45d7-bb8a-21d369d8623f/odm_meshing/odm_25dmesh.ply -logFile /var/www/data/34d695f9-4899-45d7-bb8a-21d369d8623f/odm_meshing/odm_25dmeshing_log.txt -maxVertexCount 30000 -neighbors 24 -resolution 0 -verbose
Loading point cloud... loaded 167208 points
Remove outliers... 8499 points removed
Squash point cloud to plane... Segmentation fault (core dumped)
Traceback (most recent call last):
  File "/code/run.py", line 47, in <module>
    plasm.execute(niter=1)
  File "/code/scripts/odm_meshing.py", line 105, in process
    '-resolution {resolution} {verbose}'.format(**kwargs))
  File "/code/opendm/system.py", line 34, in run
    raise Exception("Child returned {}".format(retcode))
Exception: Child returned 139

pierotofy, 2018-02-01:
I really don't know where the cause would be. Can you recover the `core` dump file and attach it here? Perhaps we'll know more once we take a look at it.

PeterSprague, 2018-02-01:
"Can you recover the core dump file and attach it here? Perhaps we'll know more once we take a look at it."
Where? 
$ whereis core
core: /usr/share/man/man5/core.5.gz

PeterSprague, 2018-02-02:
amd64 32 core 64Gb
opensfm_processes @32
opengv built from source w/o core2 flag
no fast_ortho checked
VTK v8.1.0

Success, good ortho

rerun w/ fast_ortho checked
2018-02-02 05:14:31,471 DEBUG: Undistorting image 2017_0909_115521_230.JPG
[DEBUG]   running PYTHONPATH=/code/SuperBuild/install/lib/python2.7/dist-packages /code/SuperBuild/src/opensfm/bin/opensfm export_ply --no-cameras /var/www/data/e2faa46b-2cf0-4517-8259-b4d127373831/opensfm
[DEBUG]   running PYTHONPATH=/code/SuperBuild/install/lib/python2.7/dist-packages /code/SuperBuild/src/opensfm/bin/export_bundler /var/www/data/e2faa46b-2cf0-4517-8259-b4d127373831/opensfm
[INFO]    Running ODM OpenSfM Cell - Finished
[INFO]    Running ODM Meshing Cell
[DEBUG]   Writing ODM 2.5D Mesh file in: /var/www/data/e2faa46b-2cf0-4517-8259-b4d127373831/odm_meshing/odm_25dmesh.ply
[DEBUG]   running /code/build/bin/odm_25dmeshing -inputFile /var/www/data/e2faa46b-2cf0-4517-8259-b4d127373831/opensfm/reconstruction.ply -outputFile /var/www/data/e2faa46b-2cf0-4517-8259-b4d127373831/odm_meshing/odm_25dmesh.ply -logFile /var/www/data/e2faa46b-2cf0-4517-8259-b4d127373831/odm_meshing/odm_25dmeshing_log.txt -maxVertexCount 30000 -neighbors 24 -resolution 0 -verbose
Loading point cloud... loaded 166609 points
Remove outliers... 6182 points removed
Squash point cloud to plane... Segmentation fault (core dumped)
Traceback (most recent call last):
  File "/code/run.py", line 47, in <module>
    plasm.execute(niter=1)
  File "/code/scripts/odm_meshing.py", line 105, in process
    '-resolution {resolution} {verbose}'.format(**kwargs))
  File "/code/opendm/system.py", line 34, in run
    raise Exception("Child returned {}".format(retcode))
Exception: Child returned 139

PeterSprague, 2018-02-02:
nada

2018-02-02 06:40:34,007 DEBUG: Undistorting image 2017_0909_115521_230.JPG
[DEBUG]   running PYTHONPATH=/code/SuperBuild/install/lib/python2.7/dist-packages /code/SuperBuild/src/opensfm/bin/opensfm export_ply --no-cameras /var/www/data/7c7d5211-9e70-4f3b-a7ae-53ce9d0a4ef9/opensfm
[DEBUG]   running PYTHONPATH=/code/SuperBuild/install/lib/python2.7/dist-packages /code/SuperBuild/src/opensfm/bin/export_bundler /var/www/data/7c7d5211-9e70-4f3b-a7ae-53ce9d0a4ef9/opensfm
[INFO]    Running ODM OpenSfM Cell - Finished
[INFO]    Running ODM Meshing Cell
[DEBUG]   Writing ODM 2.5D Mesh file in: /var/www/data/7c7d5211-9e70-4f3b-a7ae-53ce9d0a4ef9/odm_meshing/odm_25dmesh.ply
[DEBUG]   running /code/build/bin/odm_25dmeshing -inputFile /var/www/data/7c7d5211-9e70-4f3b-a7ae-53ce9d0a4ef9/opensfm/reconstruction.ply -outputFile /var/www/data/7c7d5211-9e70-4f3b-a7ae-53ce9d0a4ef9/odm_meshing/odm_25dmesh.ply -logFile /var/www/data/7c7d5211-9e70-4f3b-a7ae-53ce9d0a4ef9/odm_meshing/odm_25dmeshing_log.txt -maxVertexCount 30000 -neighbors 24 -resolution 0 -verbose
Loading point cloud... loaded 166854 points
Remove outliers... 7339 points removed
Squash point cloud to plane... Segmentation fault (core dumped)
Traceback (most recent call last):
  File "/code/run.py", line 47, in <module>
    plasm.execute(niter=1)
  File "/code/scripts/odm_meshing.py", line 105, in process
    '-resolution {resolution} {verbose}'.format(**kwargs))
  File "/code/opendm/system.py", line 34, in run
    raise Exception("Child returned {}".format(retcode))
Exception: Child returned 139

PeterSprague, 2018-05-22:
Atom J3455 4 core 16gb RAM
using fast-ortho
may222018
reran with complete rebuild of all images & source on Atom
used newest source -> VTK v8.1.1
successful all build including ortho
fast-ortho checked --> same error
-----------------------------------
/src/OpenSfM/bin/opensfm export_geocoords /var/www/data/1fa5a2e2-8389-4be3-9bfe-ac6e926fdee2/opensfm --transformation --proj '+units=m +no_defs=True +datum=WGS84 +proj=utm +zone=10 '
[INFO]    Running ODM OpenSfM Cell - Finished
[INFO]    Running ODM Meshing Cell
[DEBUG]   Writing ODM 2.5D Mesh file in: /var/www/data/1fa5a2e2-8389-4be3-9bfe-ac6e926fdee2/odm_meshing/odm_25dmesh.ply
[DEBUG]   running /code/build/bin/odm_25dmeshing -inputFile /var/www/data/1fa5a2e2-8389-4be3-9bfe-ac6e926fdee2/opensfm/reconstruction.ply -outputFile /var/www/data/1fa5a2e2-8389-4be3-9bfe-ac6e926fdee2/odm_meshing/odm_25dmesh.ply -logFile /var/www/data/1fa5a2e2-8389-4be3-9bfe-ac6e926fdee2/odm_meshing/odm_25dmeshing_log.txt -maxVertexCount 30000 -neighbors 24 -resolution 0 -verbose
Loading point cloud... loaded 112632 points
Remove outliers... 8261 points removed
Squash point cloud to plane... Segmentation fault (core dumped)
Traceback (most recent call last):
  File "/code/run.py", line 47, in <module>
    plasm.execute(niter=1)
  File "/code/scripts/odm_meshing.py", line 106, in process
    '-resolution {resolution} {verbose}'.format(**kwargs))
  File "/code/opendm/system.py", line 34, in run
    raise Exception("Child returned {}".format(retcode))
Exception: Child returned 139

PeterSprague, 2018-06-12:
Would the odm_25dmeshing "-resolution 0" parameter be causing it to Segfault?

pierotofy, 2018-06-12:
`0` means automatically compute a suitable resolution. You can see the exact formula in odm_25dmeshing.cpp. It's quite possible that that is the cause of the seg fault, although it's probably caused by VTK.

You'll be happy to know that we are planning to update the 2.5D meshing program and remove the VTK 7+ dependency (we are keeping the VTK version that ships prebuilt with Ubuntu, but will remove the one we are currently compiling from sources! Keep an eye on https://github.com/OpenDroneMap/OpenDroneMap/projects/2

PeterSprague, 2018-06-12:
I'm using VTK 8.1.1.  Debian Stretch is at VTK6 6.3.0
Thought VTK 8.1 had enhanced GPU and parallel processing capabilities.  Wouldn't you want that in ODM?

closed 2020-01-19 by smathermather