OpenDroneMap/ODM #755: opensfm_processes @default aka nproc crashes

PeterSprague, 2018-01-31:
Was using ODM build based off master from last October successfully with opensfm_processes @default, aka 32 cores for my workstation with my 157 block-1-east test images

Merged to latest master yesterday, and now have to adjust opensfm_processes to 16 to get a successful production of ortho.  Time to finish has been reduced slightly from 01:42 to 01:35

Wondering what has changed to induce this situation, and if there is some tweaking in the code I should be looking at?  Looking for the default settings and reading of settings.yaml code, nproc --> nproc/2 to test

dakotabenjamin, 2018-03-09:
Some things to keep in mind: 
- `--resize-to` only affects opensfm and can speed up runs
- calibrating your images helps 
- for larger datasets, `--use-hybrid-bundle-adjustment` can also speed things up

I've been seeing reports of opensfm slowdowns lately so it's definitely something we are looking into.

PeterSprague, 2018-03-19:
My newest build off ODM master that includes newest OpenCV, OpenSfM, VTK, etc built from source now doesn't choke on using -j$(nproc).
I now have a almost a 100% speed-up, going from 1:40-1:50 run times to 1:04 for the same data set and settings.

closed 2018-03-19 by PeterSprague